timvideos/HDMI2USB-litex-firmware #2: Build fails at commit 72dbf3cad04e97e4080a6ada3184f78d7a9fbc75 and later.

mithro, 2015-07-02:
```
Checking expanded design ...
WARNING:NgdBuild:440 - FF primitive 'FDPE_5' has unconnected output pin
WARNING:NgdBuild:486 - Attribute "INIT" is not allowed on symbol "ODDR" of type
   "ODDR".  This attribute will be ignored.
WARNING:NgdBuild:486 - Attribute "SRTYPE" is not allowed on symbol "ODDR" of
   type "ODDR".  This attribute will be ignored.
ERROR:NgdBuild:604 - logical block 'ODDR' with type 'ODDR' could not be
   resolved. A pin name misspelling can cause this, a missing edif or ngc file,
   case mismatch between the block name and the edif or ngc file name, or the
   misspelling of a type name. Symbol 'ODDR' is not supported in target
   'spartan6'.

Partition Implementation Status
-------------------------------

  No Partitions were found in this design.

-------------------------------

NGDBUILD Design Results Summary:
  Number of errors:     1
  Number of warnings:   3

Total REAL time to NGDBUILD completion:  6 sec
Total CPU time to NGDBUILD completion:   6 sec

One or more errors were found during NGDBUILD.  No NGD file will be written.

Writing NGDBUILD log file "atlys-minisoc-atlys.bld"...
Traceback (most recent call last):
  File "./make.py", line 198, in <module>
    vns = platform.build(soc, build_name=build_name, **build_kwargs)
  File "/usr/local/google/home/tansell/foss/m-labs/migen/mibuild/xilinx/platform.py", line 26, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/usr/local/google/home/tansell/foss/m-labs/migen/mibuild/xilinx/ise.py", line 208, in build
    self.map_opt, self.par_opt)
  File "/usr/local/google/home/tansell/foss/m-labs/migen/mibuild/xilinx/ise.py", line 136, in _run_ise
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

mithro, 2015-07-02:
Version 8853e2cb7586bcb5e167d403c55087132c064e43 builds for me but b5283923e6a9f49a313a71cde21b1bfe61c5c25b fails with the same error.

enjoy-digital, 2015-07-02:
Already fixed in Migen.

closed 2015-07-02 by enjoy-digital