timvideos/HDMI2USB-litex-firmware #6: Design does not meet timing

shenki, 2015-07-06:
ISE 14.7 with commit 0132a287cf9da81294f97aefef2439acd1ccb760

https://github.com/m-labs/misoc/commit/e011f9378fc971749dbcc65aafe5f2ca7830f22e
https://github.com/m-labs/migen/commit/73ea404380e4d16af76fd3e9e2ec1cc6eb624149

```
----------------------------------------------------------------------------------------------------------
  Constraint                                |    Check    | Worst Case |  Best Case | Timing |   Timing   
                                            |             |    Slack   | Achievable | Errors |    Score   
----------------------------------------------------------------------------------------------------------
* TS_videostreamersoc_videostreamersoc_crg_ | SETUP       |    -0.054ns|    13.764ns|       1|          54
  pll_5_ = PERIOD TIMEGRP         "videostr | HOLD        |     0.253ns|            |       0|           0
  eamersoc_videostreamersoc_crg_pll_5_"     |             |            |            |        |            
       TS_videostreamersoc_videostreamersoc |             |            |            |        |            
  _crg_clk100b / 0.75 HIGH 50%              |             |            |            |        |            
----------------------------------------------------------------------------------------------------------
* TS_videostreamersoc_clocking_clk_pix_unbu | MINPERIOD   |    -0.052ns|     1.052ns|       1|          52
  ffered_0 = PERIOD TIMEGRP         "videos | MINLOWPULSE |     6.666ns|     3.334ns|       0|           0
  treamersoc_clocking_clk_pix_unbuffered_0" |             |            |            |        |            
   TS_base50_clk HIGH         50%           |             |            |            |        |            
```

enjoy-digital, 2015-07-06:
Thanks. I'll fix that once design will be stabilized.

mithro, 2015-08-04:
This looks to be fixed now.

closed 2015-08-04 by mithro