timvideos/HDMI2USB-litex-firmware #7: Move to using git submodules to make misoc/migen versions consistent

mithro, 2015-07-06:
The version of misoc/migen could make a big difference in the build result. git has submodules (which can be a bit of a pain) but means that every commit in the master repository will use the right versions of the dependencies. A couple of small scripts can make dealing with submodules much easier too.

I'll add something to the Makefile in the near future.

enjoy-digital, 2015-07-06:
For now no need to bother with that until design is stabilized. The goal is to be able to use upstream version of Migen/MiSoC.

mithro, 2015-08-19:
Some work on this has been started here -> https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/submodules

mithro, 2015-08-19:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/ee7bb7c9dd06ef2e1c3dc95aca64a9310be9831b

mithro, 2015-10-04:
This was fixed in https://github.com/timvideos/HDMI2USB-misoc-firmware/pull/89

closed 2015-10-04 by mithro