timvideos/HDMI2USB-litex-firmware #11: DVI Sampler FIFO overflows with encoder and framebuffer enabled on 1024x768 or 720p modes

mithro, 2015-08-04:
@xfxf can add more details about his tests.

xfxf, 2015-08-10:
Tested this on two seperate monitors.  Both outputting from a Mac (Retina Macbook Pro) at both 1024x768 and 720p.   Example:

```[FLTERM] Starting...
dvi_in: 0x0 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 0x0 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
Available commands:
h               - this message
l               - list resolutions
0-9             - set resolution
D/d             - enable/disable DVI debug
E/e             - enable/disable Encoder
F/f             - enable/disable Framebuffer
p               - dump plls configuration
r               - reboot CPU
S/s             - enable/disable system status
v               - show gateware/firmware versions

dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
framebuffer is ON
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
dvi_in: 1024x768 | dvi_out: 1024x768 | encoder: OFF
Encoder is ON
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
[... 382 lines cut ...]
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
framebuffer is OFF
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvisampler: FIFO overflow
dvi_in: 1024x768 | dvi_out: OFF | encoder: 39 fps, q: 50
dvi_in: 1024x768 | dvi_out: OFF | encoder: 40 fps, q: 50
make: *** [load-lm32-firmware] Interrupt
```

enjoy-digital, 2015-08-10:
Current DDR controller cannot handle 3 streams at 1280x720@60Hz or 1024x768@60Hz.
I'm implementing YCbCr 4:2:2 to solve that.

enjoy-digital, 2015-08-11:
Now working since commit https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/58132d847b1c684f0083af15e9092e7a88f62fe0

closed 2015-08-11 by enjoy-digital

mithro, 2015-08-11:
@enjoy-digital So we should be able to enable the frame buffer and the encoder for 720p60?

mithro, 2015-08-11:
@enjoy-digital Will 1024x768@60Hz work as well now too?

enjoy-digital, 2015-08-11:
yes both are OK :)