timvideos/HDMI2USB-litex-firmware #12: USB streaming doesn't work after using FPGALink to program Atlys board

mithro, 2015-08-04:
@xfxf can add more details about his tests.

mithro, 2015-08-04:
Current theory is that the FX2 firmware doesn't actually reset everything correctly. Thus when running after fpgalink firmware, something is not correct.

mithro, 2015-08-04:
After loading the FX2 firmware, the behaviour is that the encoder drops to 0fps.

xfxf, 2015-08-10:
Not sure what else to add.  Flash gateware, flash firmware, enable encoder (but disable framebuffer due to FIFO overflow bug described in previous issue when both are enabled).

No USB webcam is available to capture from.

This is inside of a VMWare VM, but I see the same USB devices from the host (OS X).

```
dvi_in: 1024x768 | dvi_out: OFF | encoder: 40 fps, q: 50
dvi_in: 1024x768 | dvi_out: OFF | encoder: 39 fps, q: 50
dvi_in: 1024x768 | dvi_out: OFF | encoder: 39 fps, q: 50
dvi_in: 1024x768 | dvi_out: OFF | encoder: 39 fps, q: 50
dvi_in: 1024x768 | dvi_out: OFF | encoder: 39 fps, q: 50
```
xf@timcore:~/HDMI2USB-misoc-firmware$ lsusb 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 021: ID 04e2:1410 Exar Corp. 
Bus 003 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 020: ID 1d50:602b OpenMoko, Inc. FPGALink
Bus 003 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

mithro, 2015-08-10:
Here you haven't loaded the fx2 firmware (only the lm32 firmware).

enjoy-digital, 2015-08-10:
If you see encoder's fps different from 0, then the fx2 is accepting data from the FPGA.

mithro, 2015-08-19:
This seems to be working now. Ryan can you see if you can replicate this working on your board?
[image]

mithro, 2015-08-20:
The problem appears to be when you load the fx2 firmware while the encoder is running. Once you have loaded the fx2 firmware you need to do a `encoder off` and then `encoder 75 0` to get USB streaming working again.

mithro, 2015-08-21:
I think this is fixed now. I'm assigning to @xfxf to verify that it works for him too. 

@xfxf - please close the issue if it does.

enjoy-digital, 2015-08-21:
with https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/887a4ff2923401c4a9fdcd4db2d61b10bf26e843 you should no longer need to desactivate the encoder.

closed 2015-09-09 by mithro