timvideos/HDMI2USB-litex-firmware #13: Setting the resolution to 1024x768 causes Mac's to spaz out.

mithro, 2015-08-04:
@xfxf can you describe more the behaviour here.

xfxf, 2015-08-10:
Reproduced this:
* HDMI2USB output & Mac set to 720p, attempting to output on a 1080p BenQ monitor (no output)
* Unplugged HDMI2USB, plugged into different older Dell 1280x1024 monitor, output now showing 720p (with whacky squashed aspect ratio)
* Changed HDMI2USB output to 1024x760@60hz
* Image not showing correctly on monitor (as expected)
* Try to change resolution on Mac, not showing 1024x768 as an option, only 800x600 and 720p
* Unplug HDMI2USB, plug back in again, Mac beachballs and never loads Display panel
* Even if rebooting, same, cannot ever get to Display, never outputs, Mac just beachballs

* I then removed the Mac's saved EDID data (removing /System/Library/Displays/Contents/Resources/Overrides/* on OS X 10.11)
* Plugging in HDMI2USB now results in output on monitor, outputting 1024x768 immediately

I also caused the above to occur using @mithro's portable HDMI monitor, just by trying to change resolution between 720p and 1024x768.

I'm assuming the Mac is saving EDID data first discovered off the HDMI2USB, which is changing.

mithro, 2015-08-10:
FYI @enjoy-digital - @xfxf is going to take a look at EDID generation to see what we need to do to make Mac think it is a new monitor each time.

mithro, 2015-08-10:
He'll start by figuring out what fields the Mac thinks are important in the EDID by manipulating https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/edid.c

enjoy-digital, 2015-08-10:
Strange behaviour of the MAC. This is not a bug in the firmware but a workaround to find for the MAC.

mithro, 2015-08-10:
Yeah, stupid Macs :(

mithro, 2015-08-19:
@xfxf - Any progress? :)

mithro, 2016-01-18:
I'm closing this as I don't think its still reproducible?

closed 2016-01-18 by mithro