timvideos/HDMI2USB-litex-firmware #17: Encoder frame rate seems to depend on input image contents

mithro, 2015-08-19:
When giving the encoder random noise it can drop down to as little as 8fps. On normal content it seems to regularly hit 30fps.

What is going on here? Shouldn't the encoder always take the same amount of time? I guess I don't understand JPEG enough to answer properly.

enjoy-digital, 2015-08-19:
Noisy content generates larger encoded frames. I'm guessing the limitation here is the USB but we could investigate on that.

mithro, 2015-08-19:
Maybe we add to the status message the average size of the frame produced?

mithro, 2015-08-19:
BTW The system is pipelined right? We are generating a frame while sending the previous frame up the USB?

enjoy-digital, 2015-08-19:
I'll add something to compute the throughput on the USB. The system is pipelined. When decreasing encoding quality we are able to always encode @ 30fps.

enjoy-digital, 2015-11-02:
This is an expected behaviour. Status now reports Encoder's output throughput.

closed 2015-11-02 by enjoy-digital