timvideos/HDMI2USB-litex-firmware #18: HDMI output doesn't work with some monitors

mithro, 2015-08-19:
The output from the firmware doesn't seem to be fully HDMI compatible (only DVI). This seems to cause issues with some monitors.

 * Output works with one screen by changing from the "HDMI" input to the "DVI" input (despite the screen only actually having a HDMI port).
 * Output doesn't work with a screen which only has a "HDMI" input option. It doesn't detect the signal at all.

Both boards display fine with the Digilent booting firmware.

This might be related to the information at http://hamsterworks.co.nz/mediawiki/index.php/Minimal_HDMI

mithro, 2015-08-22:
So we just came across this again today. It appears that supporting HDMI formats really needs to be implemented both for capture and output.

mithro, 2015-08-22:
We also had an issue with a Raspberry Pi being detected as an input which might be related. It would say weird resolutions like 36x5 or 1288x10

mithro, 2015-08-22:
There is potentially more information at https://www.scarabhardware.com/forums/topic/strange-sync-signals-from-hdmi/ too.

CarlFK, 2015-08-27:
DVI / vga monitor http://www.viewsonic.com/us/va2246m-led.html

Power on the Atlys, see test pattern. 

I flash the Atlys, the monitor goes into sleep, and nothing I do after that wakes it up.
I am able to see hdmi input over the usb with mplayer.

```
HDMI1 connected 1920x1080+1366+0 (0xa8) normal (normal left inverted right x axis y axis) 477mm x 268mm
	Identifier: 0x44
	Timestamp:  14344124
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      0 1
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	EDID: 
		00ffffffffffff005a632e6f01010101
		0d18010380301b782e84d5a25a52a226
		0d5054bfef80b300a940a9c095009040
		818081408100023a801871382d40582c
		4500dd0c1100001e000000ff00545350
		3134313334303233310a000000fd0032
		4b185211000a202020202020000000fc
		0056413232343620534552494553007a
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	audio: auto 
		supported: force-dvi, off, auto, on
  1920x1080 (0xa8)  148.5MHz +HSync +VSync *current +preferred
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock   67.5KHz
        v: height 1080 start 1084 end 1089 total 1125           clock   60.0Hz
  1600x1200 (0xa9)  162.0MHz +HSync +VSync
        h: width  1600 start 1664 end 1856 total 2160 skew    0 clock   75.0KHz
        v: height 1200 start 1201 end 1204 total 1250           clock   60.0Hz
  1680x1050 (0xaa)  119.0MHz +HSync -VSync
        h: width  1680 start 1728 end 1760 total 1840 skew    0 clock   64.7KHz
        v: height 1050 start 1053 end 1059 total 1080           clock   59.9Hz
  1400x1050 (0xab)  101.0MHz +HSync -VSync
        h: width  1400 start 1448 end 1480 total 1560 skew    0 clock   64.7KHz
        v: height 1050 start 1053 end 1057 total 1080           clock   59.9Hz
  1600x900 (0xac)  119.0MHz -HSync +VSync
        h: width  1600 start 1696 end 1864 total 2128 skew    0 clock   55.9KHz
        v: height  900 start  901 end  904 total  932           clock   60.0Hz
  1280x1024 (0xad)  135.0MHz +HSync +VSync
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock   80.0KHz
        v: height 1024 start 1025 end 1028 total 1066           clock   75.0Hz
  1280x1024 (0xae)  108.0MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock   64.0KHz
        v: height 1024 start 1025 end 1028 total 1066           clock   60.0Hz
  1440x900 (0xaf)   88.8MHz +HSync -VSync
        h: width  1440 start 1488 end 1520 total 1600 skew    0 clock   55.5KHz
        v: height  900 start  903 end  909 total  926           clock   59.9Hz
  1280x960 (0xb0)  108.0MHz +HSync +VSync
        h: width  1280 start 1376 end 1488 total 1800 skew    0 clock   60.0KHz
        v: height  960 start  961 end  964 total 1000           clock   60.0Hz
  1280x800 (0xb1)   71.0MHz +HSync -VSync
        h: width  1280 start 1328 end 1360 total 1440 skew    0 clock   49.3KHz
        v: height  800 start  803 end  809 total  823           clock   59.9Hz
  1152x864 (0xb2)  108.0MHz +HSync +VSync
        h: width  1152 start 1216 end 1344 total 1600 skew    0 clock   67.5KHz
        v: height  864 start  865 end  868 total  900           clock   75.0Hz
  1024x768 (0xb3)   78.8MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock   60.1KHz
        v: height  768 start  769 end  772 total  800           clock   75.1Hz
[... 26 lines cut ...]
        v: height  480 start  489 end  491 total  520           clock   72.8Hz
  640x480 (0xba)   30.2MHz -HSync -VSync
        h: width   640 start  704 end  768 total  864 skew    0 clock   35.0KHz
        v: height  480 start  483 end  486 total  525           clock   66.7Hz
  640x480 (0xbb)   25.2MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock   31.5KHz
        v: height  480 start  490 end  492 total  525           clock   60.0Hz
  720x400 (0xbc)   28.3MHz -HSync +VSync
        h: width   720 start  738 end  846 total  900 skew    0 clock   31.5KHz
        v: height  400 start  412 end  414 total  449           clock   70.1Hz
DP1 disconnected (normal left inverted right x axis y axis)
	Identifier: 0x45
	Timestamp:  14344124
	Subpixel:   unknown
	Clones:    
	CRTCs:      0 1
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	audio: auto 
		supported: force-dvi, off, auto, on
```

mithro, 2015-08-27:
So, we tested two resolution modes with Carl's monitor. 800x600 and 1024x768. 
```
Setting video mode to 800x600 @56Hz
Setting video mode to 1024x768 @60Hz
```

I do notice that neither of these resolutions are supported by the EDID dump that Carl provided. I think this indicates that we need to get the EDID dumping from https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/35 working and warn if;
 * A monitor is connected which doesn't support the resolution the system is set to.
 * A resolution is set to a mode the connected monitors don't support.

I'm going to open a new github issue for that and dump the information in it.

enjoy-digital, 2015-08-27:
HSYNC and VSYNC were not generated on the right channel (red instead of blue), so this was not DVI compliant. https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/5eabb443b53c60db74caaca4677a40cf8f82346d fixes that and I managed to get video working on a monitor that was not working before this fix. Still not HDMI compliant, but maybe enough for now.

enjoy-digital, 2015-08-27:
BTW, Raspberry Pi is working fine here as a video source. It can also be a timing issue on some P&R.

enjoy-digital, 2015-09-01:
mithro, CarlFK: do you still have troubles with monitor output since https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/5eabb443b53c60db74caaca4677a40cf8f82346d ?

CarlFK, 2015-09-01:
> 5eabb44

tested with :
 Viewsonic DVI monitor that was having problems,
hp dvi
hp hdmi
all 3 are working well.

mithro, 2015-09-11:
I think this issue is resolved now?

We have a seperate issue open regarding the EDID negotiation of output @ https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/37

closed 2015-09-11 by mithro