timvideos/HDMI2USB-litex-firmware #20: Get better performance with encoder and quarter rate DDR phy

enjoy-digital, 2015-08-21:
We have to reduce system clock to use the quarter rate DDR phy. This improves DDR bandwidth but also decreases JPEG encoder performance. We should maybe run the encoder in another clock domain.

mithro, 2015-09-11:
This is also finished right?

mithro, 2015-10-01:
I believe this is dependent on the work on litedram @ https://github.com/enjoy-digital/litedram ?

enjoy-digital, 2015-10-01:
No, that's more related to changing clock domain of the encoder.

mithro, 2016-01-20:
Adding some thoughts from my investigation of this;
 * The encoder uses a SyncFIFO on the output - see https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/gateware/encoder/__init__.py#L94
 * The USB Streamer contains an AsyncFIFO - see https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/gateware/streamer/__init__.py#L89
 * The encoder is directly connected to the streamer in the targets - see https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/targets/opsis_hdmi2usb.py#L76
 * The encoder is feed from the DDR using the EncoderReader object which uses a DMA - see https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/gateware/encoder/__init__.py#L25
 * The encoder needs chroma_upsampling - see https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/gateware/encoder/__init__.py#L81

Do we just need to put a CDC object (such as an ASyncFIFO) between the EncoderReader and the Encoder modules and then run the Encoder as fast as it will go?

enjoy-digital, 2016-01-20:
Yes, we need to change clock domain for the encoder. Changing clock domain on a stream is easy (AsyncFIFO) but we also need to change clock domains for control (wishbone), I haven't found time to work on that for now... but we'll change that.

enjoy-digital, 2016-01-23:
It's working on my fork (system at 50MHz, encoder at 100Mhz).
We should now have better performance on the opsis than the atlys.
I also added a comment to configure the fps we are targeting.

Still some cleanup to do, I'll merge that soon.

mithro, 2016-01-24:
I've done a bunch of cleanup of your branch at https://github.com/mithro/HDMI2USB-misoc-firmware/tree/florent

mithro, 2017-03-19:
This should be done now.

closed 2017-03-19 by mithro