timvideos/HDMI2USB-litex-firmware #21: Add I2C EEPROM gateware and software

enjoy-digital, 2015-08-21:
I2C Eeprom stores MAC address. We have to implement gateware and software support to access the EEPROM.

mithro, 2015-08-21:
We also plan to store other manufacturing information in the EEPROM such as
the git revision of the PCB.

It would be good to support dumping it out the UART.

enjoy-digital, 2015-08-21:
gateware: https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/7ee9cdaf023cfd5997ea620d5405c64805255370
software: https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/34e8d37ee3685df65ffed18ac1853981839e5019

it needs tests and the eeprom protocol on top of that.

mithro, 2015-08-21:
The EEPROM on the Numato Opsis is a 24AA02E48 - http://www.microchip.com/wwwproducts/Devices.aspx?product=24AA02E48

> The Microchip Technology Inc. 24AA02E48 is a 2 Kbit Electrically Erasable PROM. The device is organized as two blocks of 128 x 8-bit memory with a 2-wire serial interface.
> The 24AA02XEXX is programmed at the factory with a globally unique node address stored in the upper half of the array and permanently write-protected. The remaining 1,024 bits are available for application use.

mithro, 2015-08-21:
Verilog models for the EEPROM can be found at http://ww1.microchip.com/downloads/en/DeviceDoc/24xx02_Verilog_Model.zip

Sample code for an 8051 processor can be found at http://ww1.microchip.com/downloads/en/AppNotes/01195b.pdf and http://ww1.microchip.com/downloads/en/AppNotes/AN1195%20Source%20Code.zip

mithro, 2015-10-28:
I have FX2 firmware which reads the MAC address in the EEPROM and patches it into the usb-serial adapter https://github.com/mithro/fx2lib/commit/cd7e61c03d9b69580e6d4cb8394fb0bc1159e7ed#diff-030c9487adb525e6b1e25c4c56bfee71R119

mithro, 2015-10-28:
Cleaner implementation at https://github.com/mithro/HDMI2USB-misoc-firmware/commit/fbdd64bd516be9c74d751675f2394c273a2a1eb9

mithro, 2015-10-28:
@fallen

mithro, 2016-01-10:
There is now a command to dump the Opsis EEPROM at https://github.com/mithro/HDMI2USB-misoc-firmware/commit/d11c9e9bbee638baf062852e0fb3fce982782819