timvideos/HDMI2USB-litex-firmware #23: Make the HDMI2Ethernet SoC stream proper RTP packets that can be received via gstreamer

mithro, 2015-08-21:


enjoy-digital, 2015-08-21:
This can be used as a basis:
https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/hdl/streamer/rtp.py

I'm not well aware of RTP, so I think some metadata are missing (sync, ect...)

mithro, 2015-08-21:
FYI A summer of code student looked at doing RTP stuff as part of GSoC. His blog can be found at http://hdmi2ethernet.blogspot.de/

There is a bunch of information at
 * http://hdmi2ethernet.blogspot.de/2014/05/tools-for-rtp-packet-creation.html
 * http://hdmi2ethernet.blogspot.de/2014/05/rtp-packet-crafting-using-dpkt-180.html
 * http://hdmi2ethernet.blogspot.de/2014/07/debugging-gstreamer-and-testing.html

mithro, 2016-03-27:
@aomtoku - Please take a look at this github issue.