timvideos/HDMI2USB-litex-firmware #30: Support Rohit's vmodvga board in the firmware

mithro, 2015-08-21:
https://github.com/timvideos/HDMI2USB-vmodvga

rohit91, 2015-08-21:
Coincidence! I was also thinking of porting it to misoc. Misoc & migen is
new and intriguing to me! Fascinating stuff! Only if I get the VGA capture
stable! @enjoy-digital might provide much-needed help/guidance/suggestion
in this!

enjoy-digital, 2015-08-21:
Feel free to

rohit91, 2015-08-21:
@enjoy-digital , Sure will do! Thanks!

mithro, 2015-10-01:
@rohit91 How's it going? :)

mithro, 2015-10-24:
@rohit91 ping?

rohit91, 2015-10-24:
Current progress: https://asciinema.org/a/82naz6l6q3pupejkqx0cftcml
I2C is working good. AD9984A Initialization is getting properly done! 

ToDo: Data capture from AD9984A. Still have to write code for that!

rohit91, 2015-10-24:
@mithro there is also a problem. Using Ethernet (@2.5 Volts) along with VHDCI (@3.3 Volts) results in IO-Standard voltage conflict, since both Ethernet and VHDCI are on same bank, so different voltages is not possible.

We would have to redesign the VGA board with AD9984A's Output-pins' Voltage supply as 2.5V to make it work along with Ethernet. Would require change of 3.3V regulator to 2.5V regulator and, I guess, nothing else.

mithro, 2015-10-26:
@rohit91 Great to see the progress. Want to push your code somewhere so we can take a look / help figure out the problems.

mithro, 2015-10-26:
For now, we can just disable Ethernet right?

rohit91, 2015-10-26:
@mithro Yes, that is how I'm using it for now :) Build fails on adding Ethernet core!

rohit91, 2015-10-26:
Code repo: https://github.com/rohit91/HDMI2USB-vmodvga-misoc

sealne, 2015-10-27:
presumably though for the Opsis TOFE board we want to switch to 2v5?

mithro, 2015-10-27:
The TOFE board needs to work at 3v3 level. The Ethernet in connected
differently to the Atlys.

mithro, 2016-01-10:
How is it going @rohit91 - It looked like you almost have it working?

rohit91, 2016-01-11:
I'm at the last stage. 2 days back, I tried a test capture. There was some semblance of working, but the output was messy. I will have to look where might be the problem.

rohit91, 2016-01-12:
Yay! A awesome stable capture! It was just one-line fix! :joy: 
Commit: rohit91/HDMI2USB-vmodvga-misoc@f21d8569049fbeeb3d5f08314c716819cf345d4e

Actual capture pics (all of them): https://drive.google.com/folderview?id=0B7Sp1NskALPdMjdQc1pBRExpZ1U&usp=sharing

enjoy-digital, 2016-01-12:
Congrats!

xfxf, 2016-01-12:
@rohit91 I have one of your VGA boards here if you want me to test it.  That said I can see you have cables jumpered/soldered to your board in your pic?

rohit91, 2016-01-13:
@enjoy-digital Thanks! :) But most of the heavy-lifting was already done by you! To quote Newton "If I have seen further, it is by standing on the shoulders of giants"

@xfxf Hi Ryan! :) You can try with your board without any modification. Main thing to test for is image capture stability without screen going blank. Expect getting output having very saturated colors. I'll have to fiddle with ADC's gain registers to correct those. Also, there might be random white dots on capture output. I'll look into those too. They are either due to AD9984A's incorrect register values or PCB noises (due to design or my poor hand soldering)

Just try the bit-file. FX2 firmware is not needed, since encoder is not present in the bit-file. 
In HDMI2USB console, use `help` command, and you would see `vga_switch on` listed there. Use that to switch on the VGA capture. Output would be on HDMI_OUT0 (near DC Power jack of Atlys). `status on` command should show 806 HSYNCs per VSYNC for correct capture. Also before capturing, please make sure that the VGA source (laptop etc) is outputting 1024x768@60p . I've yet to add support for other resolutions! Please let me know of any issues! Thanks! :)
Bitstream file: https://drive.google.com/file/d/0B7Sp1NskALPdbGdWdkpYLW9rNEE/view?usp=sharing

mithro, 2016-03-01:
@rohit91 ping? We should get this merged ASAP.

rohit91, 2016-03-01:
@mithro Yeah. I'll look into it tonight and get it merged in 1-2 days max.

mithro, 2016-08-10:
Gateware support is in #215