timvideos/HDMI2USB-litex-firmware #35: Dump EDID from connected monitor

mithro, 2015-08-21:
It would be good to have a command which dumped the raw EDID from the monitor connected to an TX port.

fallen, 2015-11-30:
Please give this a try: https://github.com/fallen/HDMI2USB-misoc-firmware/tree/hdmi_out_edid_dump
* Boot
* Connect a monitor to HDMI out port 
* in the serial command line interface enter the following command: 

`debug edid 0` for hdmi_out0 
`debug edid 1` for hdmi_out1

If it outputs something starting with `00 FF FF FF FF FF FF 00` it should be your monitor EDID.
You can copy paste it to http://www.edidreader.com/ to check the validity of the EDID dump.

enjoy-digital, 2015-11-30:
Thanks, I'll test in on the Opsis tomorrow (I can't test on the Atlys since
I damaged mine plugging 12V in it (instead of 5V...))

Can you keep i2c.c untouched and create specific hdmi_out0.c and
hdmi_out1.c files? (We will avoid duplicating things later as it's was done
on Mixxeo:
https://github.com/m-labs/mixxeo-soc/blob/master/software/Makefile#L49)

enjoy-digital, 2015-12-01:
It works fine here with the Opsis:

Monitor connected to output0
```
HDMI2USB>debug edid 0
00 FF FF FF FF FF FF 00 26 CD 1A 56 01 01 01 01
00 15 01 03 81 30 1B 78 2A 94 A5 A4 5A 54 A1 27
0E 50 54 BF EF 80 B3 00 A9 40 95 00 81 40 81 80
95 0F 71 4F D1 C0 02 3A 80 18 71 38 2D 40 58 2C
45 00 DD 0C 11 00 00 1E 00 00 00 FF 00 30 0A 20
20 20 20 20 20 20 20 20 20 20 00 00 00 FD 00 37
4C 1E 53 11 00 0A 20 20 20 20 20 20 00 00 00 FC
00 50 4C 32 32 37 33 48 44 53 0A 20 20 20 01 1D
HDMI2USB>debug edid 0
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
```

Monitor connected to output1
```
HDMI2USB>debug edid 0
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
HDMI2USB>debug edid 1
00 FF FF FF FF FF FF 00 26 CD 1A 56 01 01 01 01
00 15 01 03 81 30 1B 78 2A 94 A5 A4 5A 54 A1 27
0E 50 54 BF EF 80 B3 00 A9 40 95 00 81 40 81 80
95 0F 71 4F D1 C0 02 3A 80 18 71 38 2D 40 58 2C
45 00 DD 0C 11 00 00 1E 00 00 00 FF 00 30 0A 20
20 20 20 20 20 20 20 20 20 20 00 00 00 FD 00 37
4C 1E 53 11 00 0A 20 20 20 20 20 20 00 00 00 FC
00 50 4C 32 32 37 33 48 44 53 0A 20 20 20 01 1D
```
I also checked with www.edidreader.com.

Can you just create 2 separate hdmi_out0.c, hdmi_out1.c files and integrate the i2c code in it?
Then you can send a pull request or a patch.

Thanks

fallen, 2015-12-01:
Thanks for the test and the review, I updated the branch to create hdmi_out{0,1}.{h,c} like you said.

fallen, 2015-12-02:
I added support for extended EDID and rebased the pull request to fix the build for "base" targets (where there is no hdmi_out0/1)