timvideos/HDMI2USB-litex-firmware #38: laptop not seeing valid display device

CarlFK, 2015-09-03:
laptop I have been using for testing:
Acer v0.3108 Aspire 1410
hostname: negk

gateware revision: a4808ace
firmware revision: a4808ace, built Aug 28 2015 00:39:12

I plug it in, fumble around with video_mods and suddenly it starts working and I don't know what did it.  

working: laptop sees hdmi device, sends desktop video,  video is displayed on one of the hdmi outputs.  I have had both hdmi and encoder sending data to usb.

Today I am not able to get it to see a display device.

laptop hdmi out hooked to Atlys hdmi1 in.
nothing connected to hdmi0 in or either hdmi out.

both USB hooked to dc10b. (not relevant, I hope)

```
hdmi_in0:  0x0
hdmi_in1:  0x0
hdmi_out0: 1280x720 from hdmi_in0
hdmi_out1: 1280x720 from hdmi_in0
encoder: off
ddr: read: 1761Mbps  write:    0Mbps  all: 1761Mbps
```
```
HDMI1 disconnected (normal left inverted right x axis y axis)
```
```
[11273.184291] [drm:intel_tv_detect], [CONNECTOR:25:SVIDEO-1] force=0
[11274.559525] [drm:i965_irq_handler], hotplug event received, stat 0x38040000
[11274.559545] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[11274.559607] [drm:i915_hotplug_work_func], running encoder hotplug functions
[11274.559615] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[11274.559621] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[11274.559630] [drm:intel_hdmi_detect], [CONNECTOR:14:HDMI-A-1]
[11274.560151] [drm:gmbus_xfer], GMBUS [i915 gmbus dpb] NAK for addr: 0050 w(1)
[11274.560161] [drm:gmbus_xfer], GMBUS [i915 gmbus dpb] NAK on first message, retry
[11274.564120] [drm:gmbus_xfer], GMBUS [i915 gmbus dpb] NAK for addr: 0050 w(1)
[11274.564128] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[11274.564137] [drm:intel_dp_detect], [CONNECTOR:18:DP-1]
[11274.566542] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x51450064
[11274.574683] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x51450064
[11274.582668] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x51450064
[11283.200299] [drm:intel_tv_detect], [CONNECTOR:25:SVIDEO-1] force=0
[11293.216297] [drm:intel_tv_detect], [CONNECTOR:25:SVIDEO-1] force=0
```
```
juser@negk:~$ lspci | grep Graphics
00:02.0 VGA compatible controller: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller (rev 07)
00:02.1 Display controller: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller (rev 07)
```

CarlFK, 2015-09-03:
pretty sure this is because the SDA jumpers were in place.  pulled them, now negk sees a display device.

CarlFK, 2015-09-03:
well.. maybe not.  jumpers still removed, un/plugged hdmi cable, now negk does not see a display device.  

power cycled negk, Atlys, dc10, reloaded, negk does not see a display device.

mithro, 2015-09-03:
I think this confirms we need much better debugging of when EDID is
accessed (IE messages up the UART as the EDID state machine transitions
occur). This will allow us to correlate the information we are getting in
dmesg.

I would also think it would be a good idea to permanently connect liteScope
to the EDID (and on Opsis, the HPD) lines to record traces of these pins
until we are 100% certain things are solid. Probably need a way to use
flterm to dump the liteScope recordings via the console UART too.

enjoy-digital, 2015-09-04:
I'll add something in the gateware to allow EDID debugging.

mithro, 2015-09-04:
See https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/39 and https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/33

enjoy-digital, 2015-09-09:
If your issue is similar to to the one I had with a computer, this should solve it:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/9bb88b8bba0052b9ec665e5a1294801b787d7357

CarlFK, 2015-09-11:
negk laptop and all other problems working fine now.

closed 2015-09-11 by CarlFK