timvideos/HDMI2USB-litex-firmware #41: Create a HDMI2USB GUI controller

mithro, 2015-09-08:
Probably best to start with python and something like pygtk or pyqt. You'll probably want to use pyserial for talking to the HDMI2USB board.

UI could look something like this;
```
 /----------------------------------------------\
 | HDMI2USB Controller                  [X][_]  |
 +----------------------------------------------+
 | Input 1:  1024x768 @ 60 Hz                   |
 +----------------------------------------------+
 | Input 2:  1024x768 @ 60 Hz                   |
 +----------------------------------------------+
 | Output 1                                     |
 | +---------+ +---------+ +---------+ +-----+  |
 | | Input 1 | | Input 2 | | Pattern | | Off |  |
 | +---------+ +---------+ +---------+ +-----+  |
 +----------------------------------------------+
 | Output 1                                     |
 | +---------+ +---------+ +---------+ +-----+  |
 | | Input 1 | | Input 2 | | Pattern | | Off |  |
 | +---------+ +---------+ +---------+ +-----+  |
 +----------------------------------------------+
 | USB Capture                                  |
 | +---------+ +---------+ +---------+ +-----+  |
 | | Input 1 | | Input 2 | | Pattern | | Off |  |
 | +---------+ +---------+ +---------+ +-----+  |
 +----------------------------------------------+
 | Debug Terminal                               |
 | +------------------------------------------+ |
 | | HDMI2USB>                                | |
 | |                                          | |
 | +------------------------------------------+ |
 +----------------------------------------------+
```

shenki, 2015-09-09:
I'm keen to take this on.

What firmware do I want on my Atlys board to test?

Where is the protocol defined?

mithro, 2015-09-09:
No protocol is defined except in the C code at https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/ci.c

Follow instructions at https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/scripts to get firmware building set up.

enjoy-digital, 2015-09-09:
shenki; if you don't have time setting up the environment, you can use this for now: http://enjoy-digital.fr/open_hardware/hdmi2usb/atlys_hdmi2usb_gateware_2015_09_08.tar.gz

it seems mithro is setting up travis-ci so we will be able to provide prebuilt firmwares soon.

mithro, 2015-11-07:
@xfxf has been working on this issue, so I'm assigning it to him.

My version of this code can be found at https://gist.github.com/mithro/0717f9a0b39a538e1e6d - I don't know where Ryan is keeping his code currently.

xfxf, 2015-12-06:
About to start hacking on this again.
I'd like to abstract out the HDMI2USB interface to something more agnostic as a library, so we'll need to figure out an appropriate interface/protocol.  Will start prototyping something, expect it'll probably end up changing wildly but will require some direction on this.

xfxf, 2015-12-06:
@mithro @shenki I recall having issues a month ago with trying to read from the HDMI2USB serial interface, which @mithro identified as some sort of bug in the driver.  Was this solved?

mithro, 2015-12-06:
To be clear, it is not the "HDMI2USB serial interface" which has problems. It is the driver for the Exar USB UART on the Atlys board -- this isn't used on the Opsis and will eventually not used on the Atlys too.

xfxf, 2015-12-06:
right, thanks for the clarification - in any case it was a blocker for making sane progress on this for me a month ago. is there a workaround/fix?

mithro, 2015-12-06:
The workaround was the code I posted (which isn't 100% reliable, but should get you going).

@shenki was also on getting a rewritten kernel driver upstream.

xfxf, 2015-12-06:
The code above did not work for me at the time; same issues trying to read from the device.

mithro, 2015-12-06:
@xfxf - Please give it a go and report what actually happening + explain what is not happening.

mithro, 2016-03-27:
Ping!

mithro, 2017-08-10:
@xfxf has started something here -> https://github.com/xfxf/hdmi2usbmon/blob/master/simple_switch_ui.py

@xfxf mentioned;
> the tk version works for basic use, started rewriting using toga, haven't finished
> there's a later version than that, but it's on a device not near me atm
> iirc that version lets you choose inputs/outputs, and dumps debug output into a text frame, not much more