timvideos/HDMI2USB-litex-firmware #42: BOARD=base no longer builds

mithro, 2015-09-09:
```
+ make lm32-firmware
cd build/misoc && python3 make.py -X ../.. -t atlys_base -Ot firmware_filename ../../firmware/lm32/firmware.bin -Op programmer impact build-headers
                __  ___  _   ____     _____
               /  |/  / (_) / __/__  / ___/
              / /|_/ / / / _\ \/ _ \/ /__
             /_/  /_/ /_/ /___/\___/\___/
a high performance and small footprint SoC based on Migen
====== Building for: ======
Platform:  atlys
Target:    atlys_base
Subtarget: MiniSoC
CPU type:  lm32
===========================
make -C firmware/lm32 all
make[1]: Entering directory `/home/travis/build/timvideos/HDMI2USB-misoc-firmware/firmware/lm32'
 CC       isr.o
isr.c: In function ‘isr’:
isr.c:17:18: error: ‘HDMI_IN0_INTERRUPT’ undeclared (first use in this function)
  if(irqs & (1 << HDMI_IN0_INTERRUPT))
                  ^
isr.c:17:18: note: each undeclared identifier is reported only once for each function it appears in
isr.c:19:18: error: ‘HDMI_IN1_INTERRUPT’ undeclared (first use in this function)
  if(irqs & (1 << HDMI_IN1_INTERRUPT))
                  ^
make[1]: *** [isr.o] Error 1
make[1]: Leaving directory `/home/travis/build/timvideos/HDMI2USB-misoc-firmware/firmware/lm32'
make: *** [lm32-firmware] Error 2
The command "/trusty/run.py $PWD/.travis-run.sh" exited with 2.
```

enjoy-digital, 2015-09-09:
yes, I was aware of that, we have to fix the Makefile to avoid building lm32-firmware when base target is built.

enjoy-digital, 2015-09-11:
fixed with https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/3cfac0344262989617a9a3036e54f3c8266bab71

PR #44 (2015-09-12): Adding base back into Travis-CI
Fixes #42.
closed 2015-09-11 by enjoy-digital