timvideos/HDMI2USB-litex-firmware #47: Replace VHDL jpeg encoder with Verilog version

mithro, 2015-09-13:
Replace the jpeg encoder at https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/hdl/encoder/vhdl with cfelton's Verilog version at https://github.com/cfelton/test_jpeg

If https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/46 is also completed then the firmware is completely verilog.

enjoy-digital, 2015-09-13:
I tried to use verilog version of the encoder and had issue at synthesis.It should not  be that hard to fix, but that has to be fixed.

Also, some part of the code in the original version are too complicated and not optimized. I've done optimization on the code that have to be applied on the verilog version if we integrate it.

BUF_FIFO should also be refactored to reduce block ram usage. It's not mandatory to store 16 entires lines when changing the order the access to the DRAM are done. We can only store 2*16 DRAM bursts.

I'm wondering if the best solution wouldn't be to rewrite the encoder with migen.

mithro, 2015-09-13:
Rewriting it in migen would be definitely another option.