timvideos/HDMI2USB-litex-firmware #54: HDMI2USB target is broken on the Opsis

mithro, 2015-09-14:
See https://travis-ci.org/timvideos/HDMI2USB-misoc-firmware/jobs/80203042
```
Release 14.7 - ngdbuild P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.

Command Line:
/home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/Xilinx/opt/Xilinx/14.
7/ISE_DS/ISE/bin/lin64/unwrapped/ngdbuild -uc
opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf opsis_hdmi2usb-hdmi2usbsoc-opsis.ngc
opsis_hdmi2usb-hdmi2usbsoc-opsis.ngd

Reading NGO file
"/home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/misoc/build/opsis_hd
mi2usb-hdmi2usbsoc-opsis.ngc" ...
Gathering constraint information from source properties...
Done.

Annotating constraints to design from ucf file
"opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf" ...
WARNING:NgdBuild - The value of SIM_DEVICE on instance 'PLL_ADV_1' of type
   PLL_ADV has been changed from 'VIRTEX5' to 'SPARTAN6' to correct
   post-ngdbuild and timing simulation for this primitive.  In order for
   functional simulation to be correct, the value of SIM_DEVICE should be
   changed in this same manner in the source netlist or constraint file.
WARNING:NgdBuild - The value of SIM_DEVICE on instance 'PLL_ADV_2' of type
   PLL_ADV has been changed from 'VIRTEX5' to 'SPARTAN6' to correct
   post-ngdbuild and timing simulation for this primitive.  In order for
   functional simulation to be correct, the value of SIM_DEVICE should be
   changed in this same manner in the source netlist or constraint file.
WARNING:NgdBuild - The value of SIM_DEVICE on instance 'PLL_ADV_3' of type
   PLL_ADV has been changed from 'VIRTEX5' to 'SPARTAN6' to correct
   post-ngdbuild and timing simulation for this primitive.  In order for
   functional simulation to be correct, the value of SIM_DEVICE should be
   changed in this same manner in the source netlist or constraint file.
Resolving constraint associations...
Checking Constraint Associations...
WARNING:ConstraintSystem - TNM : GRPsys_clk was distributed to a DCM but new TNM
   constraints were not derived. This TNM is used in the following user groups
   or specifications:
   <TIMESPEC "TSise_sucks7" = FROM "GRPpix0_clk" TO "GRPsys_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(145)]
   <TIMESPEC "TSise_sucks8" = FROM "GRPsys_clk" TO "GRPpix0_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(146)]
   <TIMESPEC "TSise_sucks9" = FROM "GRPpix1_clk" TO "GRPsys_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(147)]
   <TIMESPEC "TSise_sucks10" = FROM "GRPsys_clk" TO "GRPpix1_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(148)]
   <TIMESPEC "TSise_sucks11" = FROM "GRPusb_clk" TO "GRPsys_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(153)]
   <TIMESPEC "TSise_sucks12" = FROM "GRPsys_clk" TO "GRPusb_clk" TIG;>
   [opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf(154)]

INFO:ConstraintSystem:178 - TNM 'GRPbase50_clk', used in period specification
   'TSbase50_clk', was traced into DCM_CLKGEN instance DCM_CLKGEN_1. The
   following new TNM groups and period specifications were generated at the
   DCM_CLKGEN output(s): 
   CLKFX: <TIMESPEC TS_hdmi_out0_clocking_clk_pix_unbuffered = PERIOD
   "hdmi_out0_clocking_clk_pix_unbuffered" TSbase50_clk HIGH 50%>

INFO:ConstraintSystem:178 - TNM 'GRPclk100', used in period specification
   'TSclk100', was traced into BUFIO2 instance BUFIO2. The following new TNM
   groups and period specifications were generated at the BUFIO2 output(s): 
   DIVCLK: <TIMESPEC TS_hdmi2usbsoc_clk100b = PERIOD "hdmi2usbsoc_clk100b"
   TSclk100 HIGH 50%>

INFO:ConstraintSystem:178 - TNM 'GRPclk100', used in period specification
   'TSclk100', was traced into DCM_CLKGEN instance DCM_CLKGEN. The following new
   TNM groups and period specifications were generated at the DCM_CLKGEN
   output(s): 
   CLKFX: <TIMESPEC TS_base50_clk = PERIOD "base50_clk" TSclk100 HIGH 50%>
[... 226 lines cut ...]
Mapping completed.
See MAP report file "opsis_hdmi2usb-hdmi2usbsoc-opsis_map.mrp" for details.
Problem encountered during the packing phase.

Design Summary
--------------
Number of errors   :   3
Number of warnings :   2
Traceback (most recent call last):
  File "make.py", line 200, in <module>
    vns = platform.build(soc, build_name=build_name, **build_kwargs)
  File "/home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/migen/mibuild/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/migen/mibuild/xilinx/ise.py", line 197, in build
    self.map_opt, self.par_opt)
  File "/home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/migen/mibuild/xilinx/ise.py", line 124, in _run_ise
    raise OSError("Subprocess failed")
OSError: Subprocess failed
make: *** [gateware] Error 1
travis_time:end:0dd82a26:start=1442227993825402124,finish=1442228663016460292,duration=669191058168
[0K
[31;1mThe command "/trusty/run.py $PWD/.travis/run.sh" exited with 2.[0m

Done. Your build exited with 1.
```

mithro, 2015-09-14:
Opsis
```
Selected Device : 6slx45tfgg484-3 

Slice Logic Utilization: 
 Number of Slice Registers:           12865  out of  54576    23%  
 Number of Slice LUTs:                24868  out of  27288    91%  
    Number used as Logic:             14991  out of  27288    54%  
    Number used as Memory:             9877  out of   6408   154% (*) 
       Number used as RAM:             9680
       Number used as SRL:              197

Slice Logic Distribution: 
 Number of LUT Flip Flop pairs used:  29331
   Number with an unused Flip Flop:   16466  out of  29331    56%  
   Number with an unused LUT:          4463  out of  29331    15%  
   Number of fully used LUT-FF pairs:  8402  out of  29331    28%  
   Number of unique control sets:       481

IO Utilization: 
 Number of IOs:                         125
 Number of bonded IOBs:                 109  out of    296    36%  

Specific Feature Utilization:
 Number of Block RAM/FIFO:              114  out of    116    98%  
    Number using Block RAM only:        114
 Number of BUFG/BUFGCTRLs:               13  out of     16    81%  
 Number of DSP48A1s:                     22  out of     58    37%  
 Number of PLL_ADVs:                      4  out of      4   100%  

WARNING:Xst:1336 -  (*) More than 100% of Device resources are used
```

Atlys
```
Device utilization summary:
---------------------------

Selected Device : 6slx45csg324-3 

Slice Logic Utilization: 
 Number of Slice Registers:           12504  out of  54576    22%  
 Number of Slice LUTs:                20768  out of  27288    76%  
    Number used as Logic:             14855  out of  27288    54%  
    Number used as Memory:             5913  out of   6408    92%  
       Number used as RAM:             5638
       Number used as SRL:              275

Slice Logic Distribution: 
 Number of LUT Flip Flop pairs used:  25181
   Number with an unused Flip Flop:   12677  out of  25181    50%  
   Number with an unused LUT:          4413  out of  25181    17%  
   Number of fully used LUT-FF pairs:  8091  out of  25181    32%  
   Number of unique control sets:       490

IO Utilization: 
 Number of IOs:                         137
 Number of bonded IOBs:                 122  out of    218    55%  

Specific Feature Utilization:
 Number of Block RAM/FIFO:              116  out of    116   100%  
    Number using Block RAM only:        116
 Number of BUFG/BUFGCTRLs:               12  out of     16    75%  
 Number of DSP48A1s:                     22  out of     58    37%  
 Number of PLL_ADVs:                      4  out of      4   100%  
```

enjoy-digital, 2015-09-14:
related to https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/53, buf_fifo is taking too much ressource. I'm trying to reduce ram usage of others peripherals to allow building opsis target.
The reason it broke it that I increased a little bit firmware_ram.

enjoy-digital, 2015-09-14:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/70d371c54dc18f8162bbdfb2cd09065b2b7e0fb4 should fix it.

mithro, 2015-09-18:
Fixed!

closed 2015-09-18 by mithro