timvideos/HDMI2USB-litex-firmware #57: Jay's whacky chrome book is still misbehaving

CarlFK, 2015-09-16:
Jay's chrome book exhibited odd behaviour before the EDID problem fix landed.
the symptoms were: its gui display manager wold show 1024x720 and 800x600.  and it would report the display model as whatever device was previously plunged in.  
trying all 10 video_modes did not seem to have any affect.
No other machine behaved like this.

Now it shows the correct device name and 1280x720 resolution, and extends the desktop (we can drag a window off the screen) but status shows:
input1:  9x0
output0: 1280x720@60Hz from input1 - shows black with a vertical row of blue dots 

This makes everything work:
video_mode 9  (same as currently set)  
input1:  1280x720

output and encoder show the expected desktop.

un/plug hdmi
input1:  9x0  ( black output)
video_mode 9 
input1:  1280x720  (expected output)

CarlFK, 2015-09-20:
input1: 0x0
hit reset, 
status
 input1:  1280x720

mithro, 2015-10-06:
This should be fixed when https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/80 is fixed.
It might also need https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/79 to be totally fixed.