timvideos/HDMI2USB-litex-firmware #66: Travis-CI should publish prebuilt firmware packages

mithro, 2015-09-19:
After something has been committed to master. Travis-CI should commit the firmware to an "unstable prebuilt binaries" repository so people can test it.

sealne, 2015-10-05:
this looks like it could be adapted?

https://eshepelyuk.github.io/2014/10/28/automate-github-pages-travisci.html

mithro, 2015-10-06:
Looks like you are making good progress on this @sealne - I'm going to mark it as "in progress" and assign it to you :)

There was also some stuff at https://github.com/timvideos/HDMI2USB-firmware-prebuilt - see the [.update.sh](https://github.com/timvideos/HDMI2USB-firmware-prebuilt/blob/master/.update.sh) script.

mithro, 2015-10-07:
This was mostly done in https://github.com/timvideos/HDMI2USB-misoc-firmware/pull/92 - we just need to figure out why the build is failing at https://travis-ci.org/timvideos/HDMI2USB-misoc-firmware/builds/84087820

mithro, 2015-10-07:
The build has now published the prebuilt firmware! See the latest commits at https://github.com/timvideos/HDMI2USB-firmware-prebuilt/commits/master
```
- Uploading built files to github.com/timvideos/HDMI2USB-firmware-prebuiltarchive/v0.0.0-396-g3b0f648/atlys/hdmi2usb
---------------------------------------------
Cloning into 'HDMI2USB-firmware-prebuilt'...
remote: Counting objects: 221, done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 221 (delta 8), reused 0 (delta 0), pack-reused 197
Receiving objects: 100% (221/221), 2.97 MiB | 0 bytes/s, done.
Resolving deltas: 100% (95/95), done.
Checking connectivity... done.
- Uploading .bit and logfile
- Uploading FX2 firmware
- Added symlink of atlys/firmware/unstable -> archive/v0.0.0-396-g3b0f648/atlys/hdmi2usb
[master 0f759a1] Travis build #214 of v0.0.0-396-g3b0f648
 4 files changed, 13158 insertions(+), 1 deletion(-)
 create mode 100644 archive/v0.0.0-396-g3b0f648/atlys/hdmi2usb/atlys_hdmi2usb-hdmi2usbsoc-atlys.bit
 create mode 100644 archive/v0.0.0-396-g3b0f648/atlys/hdmi2usb/hdmi2usb.hex
 create mode 100644 archive/v0.0.0-396-g3b0f648/atlys/hdmi2usb/output.2015_10_07.log
=============================================
```

closed 2015-10-07 by mithro