timvideos/HDMI2USB-litex-firmware #74: flcli build script

CarlFK, 2015-09-23:
This separates the "get the code" from "run make to build" 
Previously make would do lots of wget and untaring.

It paves the way to packaging.  
(11:44:16 PM) tumbleweed: so, you have 2 possible approaches here
(11:44:22 PM) tumbleweed: tar up everything into one big package, yeah
(11:44:37 PM) tumbleweed: or package each thing separately
(11:45:57 PM) CarlFK: pretty sure separately will require re-writing the Makefile(s)

I vote for one package that contains both flcli and fx2loader.

I am not exactly sure how this script fits into the packaging process.
```
#!/bin/bash -ex

sudo apt-get install \
  build-essential libreadline-dev libusb-1.0-0-dev python-yaml

mkdir makestuff; cd makestuff
git clone https://github.com/makestuff/common.git

mkdir libs; cd libs
for i in libargtable2 libbuffer libdump liberror libfpgalink libreadline libutpp libusbwrap libfx2loader;
  do 
  git clone https://github.com/makestuff/$i.git
  echo dev > $i/.branch
done

(
cd libargtable2
patch -p0 <~/paatch.sh.diff
# do what the patch just removed
/usr/bin/wget -O argtable2.tgz http://downloads.sourceforge.net/project/argtable/argtable/argtable-2.13/argtable2-13.tar.gz
tar zxf argtable2.tgz
rm argtable2.tgz
cd argtable2-13
./configure
cd src
mv *.c ../..
mv *.h ../..
mv *.def ../..
)
cd ..

mkdir apps; cd apps
git clone https://github.com/makestuff/flcli.git
git clone https://github.com/makestuff/fx2loader.git

cd flcli
make deps
cd ..

cd fx2loader
make deps
cd ..
```
paatch.sh.diff:
(removes things that are taken care of above)
```
diff --git Makefile Makefile
index ca54cd9..7313cf1 100644
--- Makefile
+++ Makefile
@@ -36,9 +36,7 @@ CONFIGS :=
 
 -include $(ROOT)/common/top.mk
 
-foo: argtable2.tgz
+foo: 
 	$(GET_ARGTABLE)
 	make dbg rel
 
-argtable2.tgz:
-	wget -O argtable2.tgz http://downloads.sourceforge.net/project/argtable/argtable/argtable-2.$(AT2_VERSION)/argtable2-$(AT2_VERSION).tar.gz
diff --git paatch.sh paatch.sh
index 27fa61a..07ab9d6 100755
--- paatch.sh
+++ paatch.sh
@@ -15,18 +15,6 @@
 # along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #
 #!/bin/bash
-rm -rf argtable2-${AT2_VERSION}
-tar zxf argtable2.tgz
-cd argtable2-${AT2_VERSION}
-if [ "$OS" != "Windows_NT" ]; then
-	./configure
-fi
-cd src
-mv *.c ../..
-mv *.h ../..
-mv *.def ../..
-cd ../..
-pwd
 
 ${PATCH} arg_int.c <<EOF
 --- argtable2-12/src/arg_int.c	2010-02-06 08:26:46.000000000 +0000
```

CarlFK, 2015-09-23:
```
juser@dc10b:~/temp/makestuff/apps/fx2loader$ sudo make install 
make: *** No rule to make target `install'.  Stop.
```
awesome.

CarlFK, 2015-09-23:
these are the binaries that need to be deployed 
```
flcli      libargtable2.so  liberror.so     libfx2loader.so
fx2loader  libbuffer.so     libfpgalink.so  libusbwrap.so
```

mithro, 2015-10-28:
This information would be better as a wiki page rather than a github issue. Maybe put the information in https://github.com/timvideos/HDMI2USB/wiki/libFPGALink ?

closed 2015-10-28 by mithro