timvideos/HDMI2USB-litex-firmware #77: DDR Memory is currently broken on the Opsis

mithro, 2015-09-26:
```
$ git describe
v0.0.0-367-gb3f0b8f
```

```
BIOS> revision
c355d61e
BIOS> memtest
Memtest failed: 258598/532736 words incorrect
BIOS> memtest
Memtest failed: 249756/532736 words incorrect
BIOS> memtest
Memtest failed: 270214/532736 words incorrect
BIOS> memtest
Memtest failed: 249513/532736 words incorrect
BIOS> 
```

enjoy-digital, 2015-10-13:
Closing it since since working here with upstream versions hdmi2usb and all submodules.
Please re-open it if you still have issues.

Note: when failing, you will probably still be able to run the design with "romboot" command.

mithro, 2015-10-14:
It appears the board is booting fine now, but if I reboot and run a memtest I get errors.

```
tansell@tansell-z620-l2:~/foss/timvideos/hdmi2usb/HDMI2USB-misoc-firmware$ make connect-lm32
third_party/misoc/tools/flterm --port /dev/ttyACM0 --speed 115200
[FLTERM] Starting...

HDMI2USB>help
Available commands:
help                           - this command
version                        - firmware/gateware version
reboot                         - reboot CPU
status <on/off>                - enable/disable status message (same with by pressing enter)

video_matrix list              - list available video sinks and sources
video_matrix connect <source>  - connect video source to video sink
                     <sink>

video_mode list                - list available video modes
video_mode <mode>              - select video mode

hdp_toggle <source>            - toggle HDP on source for EDID rescan

output0 on                     - enable output0
output0 off                    - disable output0

output1 on                     - enable output1
output1 off                    - disable output1

encoder on                     - enable encoder
encoder off                    - disable encoder
encoder quality <quality>      - select quality

debug pll                      - dump pll configuration
debug ddr                      - show DDR bandwidth

HDMI2USB>status
input0:  0x0
input1:  0x0
output0: 1280x720@60Hz from input0
output1: 1280x720@60Hz from input0
encoder: off
ddr: read: 1760Mbps  write:    0Mbps  all: 1760Mbps
HDMI2USB>reboot

MiSoC BIOS   http://m-labs.hk
(c) Copyright 2007-2014 Sebastien Bourdeauducq
Revision 8e860e3a built Oct 14 2015 16:49:37

BIOS CRC passed (c10d0f37)
Running on MiSoC rev. 8e860e3a (sysid:UN) at 50MHz
Initializing SDRAM...
Memtest failed: 271264/532736 words incorrect
Memory initialization failed
BIOS> memtest
Memtest failed: 219847/532736 words incorrect
BIOS> memtest
Memtest failed: 183676/532736 words incorrect
BIOS> 
```

enjoy-digital, 2015-10-14:
OK, can you try to stop output0 and output1 and do a reboot to see if you have the same behaviour?
We should maybe disable all DMA before doing a reboot.

mithro, 2015-10-14:
Still the same thing it appears;
```
HDMI2USB>status
input0:  0x0
input1:  0x0
output0: 1280x720@60Hz from input0
output1: 1280x720@60Hz from input0
encoder: off
ddr: read: 1757Mbps  write:    0Mbps  all: 1757Mbps
HDMI2USB>output0 off
Disabling output0
HDMI2USB>output1 off
Disabling output1
HDMI2USB>status
input0:  0x0
input1:  0x0
output0: off
output1: off
encoder: off
ddr: read:    0Mbps  write:    0Mbps  all:    0Mbps
HDMI2USB>reboot

MiSoC BIOS   http://m-labs.hk
(c) Copyright 2007-2014 Sebastien Bourdeauducq
Revision 8e860e3a built Oct 14 2015 16:49:37

BIOS CRC passed (c10d0f37)
Running on MiSoC rev. 8e860e3a (sysid:UN) at 50MHz
Initializing SDRAM...
Memtest failed: 527460/532736 words incorrect
Memory initialization failed
BIOS> memtest
Memtest failed: 524388/532736 words incorrect
BIOS> memtest
Memtest failed: 525000/532736 words incorrect
BIOS> memtest
Memtest failed: 524304/532736 words incorrect
BIOS> 
```

enjoy-digital, 2015-10-14:
Hmm OK, thanks for the test.

mithro, 2015-10-14:
There was a way to get more verbose information out of memtest, but I forget how?

enjoy-digital, 2015-10-14:
For now, you can type "romboot" to restart the firmware

sealne, 2015-10-18:
running the opsis-fixes branch:

kd@azur:~/dev/HDMI2USB-misoc-firmware-mithro$ make connect-lm32 
third_party/misoc/tools/flterm --port /dev/ttyACM0 --speed 115200
[FLTERM] Starting...

Command not found
BIOS> memtest
Memtest failed: 532480/532736 words incorrect
BIOS> 
BIOS> romboot
Executing booted program.

HDMI2USB firmware  http://timvideos.us/
Board's DNA: 00000000005c3875
Revision 8e860e3a built Oct 17 2015 17:32:40
HDMI2USB>

mithro, 2016-01-10:
I'm not seeing this issue any more.

@sealne Are you able to still reproduce this problem?

sealne, 2016-01-11:
i get no output now running memtest now

mithro, 2016-01-26:
@enjoy-digital This is back again now.....

mithro, 2016-01-26:
```
BIOS> memtest
Memtest failed: 532480/532736 words incorrect
BIOS> 
```

mithro, 2016-01-26:
Could there be a missing timing constraint somewhere. This means that some builds end up with a bad memory controller?