timvideos/HDMI2USB-litex-firmware #79: The EDID being read should notify the softcore

mithro, 2015-10-03:
On the Atlys we don't have access to the hotplug line, so we don't know when a new device is plugged in. Most devices will however read the EDID information right after being connected. Hence if we have yet to see a valid signal, the EDID being read is a good sign we should try looking for one. 

Hence the softcore needs to be notified when the EDID is read.

mithro, 2015-10-28:
This is possible a duplicate of https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/39

fallen, 2015-11-23:
It is straight forward to set ``hdmi_in0_connected`` (https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/hdmi_in0.c#L95) to 1 when the EDID is being read.

But on which condition set it back to 0? How can we detect the source has been unplugged?
Do we keep ``hdmi_in0_connected`` set to 1 regardless?

mithro, 2015-11-24:
I think we want a `hdmi_in0_edid_started` and a `hdmi_in0_edid_finished` notifications (maybe with timestamps?). Like normal interrupts, they should be triggered by the gateware and then clearable by the lm32 firmware.

fallen, 2015-11-24:
Can someone test this on Atlys board / hdmi2usb target?
https://github.com/fallen/HDMI2USB-misoc-firmware/commit/5d771720b6553665b87058ae7fbe1d077dd0a5f6