timvideos/HDMI2USB-litex-firmware #86: Get lm32 firmware running under qemu to enable testing without hardware

mithro, 2015-10-03:
It would be really awesome if we could "test" the lm32 firmware without needing the actual hardware (or maybe on travis). qemu has lm32 emulation support -- see http://lekernel.net/blog/2010/01/milkymist-qemu-support/

You'd need to figure out a way to either;
 * Emulate the HDMI2USB "hardware" in qemu
 * Mock out the HDMI2USB functions

It would be awesome for developing things like https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/81 and https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/41

enjoy-digital, 2015-10-03:
Simulations with verilator would probably be faster to set up:
- https://asciinema.org/a/17175
- https://asciinema.org/a/18224

mithro, 2015-10-07:
This would be useful for running tests on travis too.