timvideos/HDMI2USB-litex-firmware #87: Move non-build stuff out of the `build` directory (so you can rm -rf build)

mithro, 2015-10-03:
At the moment we can't just blow away the `build` directory because we end up storing things like dependencies (migen/misoc/liteeth) and toolchains (the stuff from conda / Xilinix tools) in the directory.

We should use submodules for the dependencies and store them in a third_party or 3rd directory - see https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/7

We should store the toolchains stuff in a different directory to build - no idea what to call that though.

mithro, 2015-10-06:
Things in git have been moved but the toolchains still need to be done.

CarlFK, 2015-10-06:
Can you better define toolchains?

I suspect the makestuff stuff is one of them.    I hope that will be an
apt-get install so no more checkout/build .

mithro, 2015-10-06:
Yes correct - makestuff is one, the stuff downloaded using conda (gcc+binutils) and of course the Xilinx tools.

We might end up replacing the makestuff with [openocd](http://openocd.org/) for JTAG programming. MiSoC is trying to standardize on using openocd and it is much easier to package.

mithro, 2017-03-19:
This is mostly done now.

commit fe3d8a8b (2015-10-04): Moving to using submodules.

Fixes Issue #7. Also helps fix #87.

All submodules are found under the third_party/ directory.
closed 2017-03-19 by mithro