timvideos/HDMI2USB-litex-firmware #90: `make load-lm32` should actually load the firmware and then exit

mithro, 2015-10-06:
The `make load-lm32` target connects to the lm32 serial console and then you need to reboot the currently running firmware and then make the bios do a serial boot (by pressing F6 I think?).

Maybe we can automate this by just cat the commands into the flterm command?

mithro, 2015-10-06:
@enjoy-digital Is there a way to make a "reboot serial" command which will make the bios go directly to the serialboot rather then having to hit F6?

enjoy-digital, 2015-10-13:
fixed via https://github.com/m-labs/misoc/commit/9041868291be0d55e4cf2295999f3b09f6ea8778.
You still have to type "reboot" or reset the board, but no need to interrupt the boot sequence.

mithro, 2015-10-13:
With no longer needing the other keypress we should just be able to `echo "reboot" | flterm xxxxx` to make it totally automatic?

enjoy-digital, 2015-10-13:
yes probably.