timvideos/HDMI2USB-litex-firmware #105: Pull in the "Device DNA" part from arty-soc

mithro, 2015-10-10:
https://github.com/enjoy-digital/arty-soc/commit/5bc9909c1af9ba6b186efc627f622676544d6359#commitcomment-13687669
and
https://github.com/enjoy-digital/arty-soc/blob/master/gateware/dna.py

I have patches for [openocd](http://openocd.org) which let you read the [DNA from a Spartan-6](http://openocd.zylin.com/#/c/2960/).

Also https://github.com/timvideos/HDMI2USB/issues/36

enjoy-digital, 2015-10-13:
added in https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/197d81f67fa0314e2762681d762b0463c7a0eff1. Could you check that it returns the same DNA than with openocd?

mithro, 2015-10-14:
The DNA appears to be quite different then what OpenOCD reports;
```
HDMI2USB firmware  http://timvideos.us/
Board's DNA: 000000000156121a
Revision 8e860e3a built Oct 14 2015 16:49:34
HDMI2USB>reboot
```

```
$ ../opencode-out/bin/openocd -f board/numato_opsis.cfg -c "init; xc6s_print_dna xc6s.tap; exit"                                                                                                                     
Open On-Chip Debugger 0.10.0-dev-00045-g664cfb1 (2015-10-14-17:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'usb_blaster' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Warn : incomplete ublast_vid_pid configuration
jtagspi_program
Info : usb blaster interface using libftdi
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc6s.tap tap/device found: 0x44028093 (mfg: 0x049, part: 0x4028, ver: 0x4)
DNA = 100111001101101011000101001010110101100001001000011010101 (0x139b58a56b090d5)
```

mithro, 2015-10-14:
The device DNA in the Spartan-6 is 57 bits long. Looks like [DNA currently assumes 63 bits?](https://github.com/enjoy-digital/arty-soc/blob/master/gateware/dna.py#L49)

mithro, 2015-10-14:
From *7 Series FPGAs Configuration User Guide UG470 (v1.10) June 24, 2015*
<blockquote>
The 7 series FPGA contains an embedded, 64-bit device identifier which is used to provide
a 57-bit Device DNA value. The identifier is nonvolatile, permanently programmed by
Xilinx into the FPGA, and is unchangeable making it tamper resistant. Each device is
programmed with a 57-bit DNA value that is most often unique. However, up to 32
devices within the family can contain the same DNA value. The JTAG FUSE_DNA 
command can be used to read the entire 64-bit value that is always unique. Device DNA is
composed of bits 63 to 7 of the 64-bit FUSE_DNA value.

External applications can access the Device DNA or FUSE_DNA values through the JTAG
port, and FPGA designs can access the DNA only through a Device DNA Access Port
(DNA_PORT).

The FPGA application accesses the DNA value using the Device DNA Access Port
(DNA_PORT) design primitive, shown in Figure 5-15.
</blockquote>

From *Spartan-6 FPGA Configuration User Guide UG380 (v2.7) October 29, 2014*
<blockquote>
Spartan-6 FPGAs contain an embedded, unique device identifier (device DNA). The
identifier is nonvolatile, permanently programmed into the FPGA, and is unchangeable,
making it tamper resistant.

As shown in Figure 5-14, the device DNA value is 57 bits long. The two most-significant
bits are always 1 and 0. The remaining 55 bits are unique to a specific Spartan-6 FPGA.

The FPGA's internal device identifier, plus any values shifted in on the DIN input, can be
read via the JTAG port using the private ISC_DNA command. This requires the
ISC_ENABLE to be loaded before the ISC_DNA command is issued.

Bit 56 of the identifier, shown in Figure 5-14, appears on the TDO JTAG output following
the ISC_DNA command when the device enters the Shift-DR state. The remaining Device
DNA bits and any data on the input to the register are shifted out sequentially while the
JTAG controller is left in the Shift-DR state. When this operation is complete, the
ISC_DISABLE command should be issued.
</blockquote>

mithro, 2015-10-14:
BTW The output of OpenOCD has been crosschecked with what iMPACT outputs.

enjoy-digital, 2015-10-14:
OK, I'll investigate on that.

mithro, 2015-10-28:
@fallen - This might be worth looking into when you finish the other I2C work.

enjoy-digital, 2015-10-29:
Sorry, I just did it before seeing your message... :)

mithro, 2015-10-29:
I'll check the value is the same as openocd here in a second. (Reopening and assign to me.)

enjoy-digital, 2015-11-02:
Closing this, we get the same DNA with iMPACT, OpenOCD and the code implemented in the gateware.

closed 2015-11-02 by enjoy-digital

mithro, 2015-11-02:
I confirmed this on the weekend and forgot to close it.