timvideos/HDMI2USB-litex-firmware #111: defaults for video_matrix

CarlFK, 2015-10-13:
Can these be set as the default:
```
encoder on
encoder quality 85
video_matrix connect input1 output0
video_matrix connect input1 output1
video_matrix connect input1 encoder
```
which should result in:
```
output0: 1280x720@60Hz from input1
output1: 1280x720@60Hz from input1
encoder: 1280x720 @ 0fps (0Mbps) from input1 (q: 85)
```

I would rather this be set in the firmware and not via a post setup action as currently the setup is less than rock solid and I want to keep it as simple as possible.

Should the reboot routine reset to default? 
I think that would be good.

enjoy-digital, 2015-10-13:
SInce everyone will want a different default configuration, the right way to do it is to enable the user to store a configuration in the flash and reload it at startup.

CarlFK, 2015-10-29:
the config will over ride the default.  We still need a default. Is there some reason not to use what I asked for?

commit ab04ede0 (2019-05-03): scripts: Fix upstream repository detection.

Fixes #111.
closed 2016-02-03 by mithro