timvideos/HDMI2USB-litex-firmware #113: Store configuration information in the same SPI flash that stores the gateware

mithro, 2015-10-14:
It would be good to store configuration information in the SPI flash that stores the gateware.

Both the Opsis and Atlys have similar SPI flash available that is much bigger then required for the gateware.

On the Opsis, the SPI flash will also be storing two firmware for the FX2 as well. So we need some type of "file system" I guess?

enjoy-digital, 2015-10-14:
Seems related to https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/111.
We first have to define memory map of the flash: bitstream / FX2 firmware1 / FX2 firmware2 and user data.
For user data, we can try to keep things simple (store raw data) or use a file system. If we choose to use a file system, IIRC something similar exists in ARTIQ.

mithro, 2015-10-14:
I was suggesting we spit the flash into 2 parts, bit stream and "user".

Then put a simple file system on the user section which lets us store the
fx2 firmwares, user configs and maybe a logo or two?

shenki, 2015-10-15:
Is the overhead of a filesystem worth it? Could you have "partitions", with a simple header in the first page of the flash that describes the layout?

mithro, 2015-10-15:
The first page has to be what ever Xilinx wants it to be for the FPGA.

The last page can be used for such a layout. That layout is pretty much a filesystem though :)

The simplest filesystem is just a concatenated list of; 
```
filename (null terminated string)
filesize (32bit int)
pointer to filedata (pointer to byte array // offset)
```

shenki, 2015-10-15:
Yeah. That sounds good to me.

mithro, 2015-10-28:
@fallen - FYI