timvideos/HDMI2USB-litex-firmware #114: input0 appears to be broken

xfxf, 2015-10-14:
input0 appears to be broken for myself and @CarlFK (two seperate Atlys boards) nothing plugged into it is ever seen by the HDMI2USB device, and the device doesn't see an external display.

```HDMI2USB>status
input0:  0x0
input1:  1280x720
output0: 1280x720@50Hz from input1
output1: 1280x720@50Hz from input0
```

this is with a multitude of different laptops, cameras, cables, etc.  all works perfectly on input1.  should be easily reprodcable with current firmware.

pins are all set correctly as per instructions in 'video_matrix help'

i tried stepping back to earlier firmware (about a month ago) and it still didn't work.

mithro, 2015-10-14:
@xfxf Can you include a link to the firmware versions you tested?

enjoy-digital, 2015-10-14:
Is the micro-hdmi port connected to a monitor? (should be output1 but I'm not sure...). If so, can you try input0 without the micro-hdmi port connected to a monitor?

CarlFK, 2015-10-14:
Nothing hooked to micro-hdmi.  and yes it is output1.

Thinking about this, I am not sure input0 ever worked.
I have always used input1 because it is on the same side of the Atlys as
the full sized hdmi output, so the 2 hdmi cables were next to each other.

mithro, 2015-10-17:
@xfxf @CarlFK @enjoy-digital 

I saw the discussion that Carl had gotten input1 working? http://logs.timvideos.us/%23timvideos/%23timvideos.2015-10-15.log.html#t2015-10-15T21:20:54

Sounds like there is something wrong with the jumper instructions?

mithro, 2015-10-17:
Assigning to @CarlFK for feedback.

mithro, 2015-10-28:
@xfxf can we close this?

mithro, 2015-10-28:
Sounds like there is something wrong with the jumper instructions?

enjoy-digital, 2015-10-28:
Closing this, xfxf's capture:
```text
input0:  1280x720
input1:  1280x720
output0: 1280x720@50Hz from input0
output1: 1280x720@50Hz from input0
```
so yes it's working.

mithro, 2015-10-29:
@enjoy-digital - Yeah but we didn't change any code right? Why is it working now and not before? I think it was because their jumpers were not set correctly? I'd like @xfxf or @CarlFK to help fix the documentation here.

enjoy-digital, 2015-11-02:
@CarlFK already updated the documentation with the capture I did:
https://github.com/timvideos/HDMI2USB/wiki/Cabling

I just added some indications in the platform file: https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/36b6341d97c6434bf36ecca88fcb6cd2fc3f646b

Note: on Atlys, when using Micro-D connector output (output1) and input0, since EDID lines are shared, user should first disconnect output1, let EDID transaction happen for input0 and can then reconnect output1. Without doing that, output1 will be working, but input0 will not. This is a limitation of the Atlys board.

closed 2015-11-02 by enjoy-digital