timvideos/HDMI2USB-litex-firmware #119: Create a minimal version of get-env scripts that only gets the tools required to flash

sealne, 2015-10-20:
Currently scripts/get-env-root.sh and scripts/get-env.sh (which are called by the bootstrap or manually) pull in every thing needed to build the firmware and gateware.

However not all of that is needed simply to flash the files from the prebuilt repo.

mithro, 2015-10-20:
I think we just want to integrate this into the modeswitch tool rather than
into this repository. You don't need all the gateware / firmware code to
flash a board either.

closed 2015-10-20 by sealne

mithro, 2015-10-24:
Just a FYI for everyone - the mode-switch tool is now available at https://github.com/timvideos/HDMI2USB-mode-switch

It doesn't do the firmware download yet, love some help getting that working.