timvideos/HDMI2USB-litex-firmware #120: Bootstrap fails

shenki, 2015-10-21:
Ubuntu Wily (15.10)

```
+ set -e
+ echo '             This script is: /home/shenki/tmp/HDMI2USB-misoc-firmware/scripts/get-env.sh'
             This script is: /home/shenki/tmp/HDMI2USB-misoc-firmware/scripts/get-env.sh
+ echo '         Firmware directory: /home/shenki/tmp/HDMI2USB-misoc-firmware'
         Firmware directory: /home/shenki/tmp/HDMI2USB-misoc-firmware
+ echo '         Build directory is: /home/shenki/tmp/HDMI2USB-misoc-firmware/build'
         Build directory is: /home/shenki/tmp/HDMI2USB-misoc-firmware/build
+ echo '     3rd party directory is: /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party'
     3rd party directory is: /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party
+ '[' '!' -d /home/shenki/tmp/HDMI2USB-misoc-firmware/build ']'
++ tempfile
+ XILINX_PASSPHRASE_FILE=/tmp/fileurlDBq
+ trap 'rm -f -- '\''/tmp/fileurlDBq'\''' EXIT
+ set +x
+ '[' -f /tmp/fileurlDBq ']'
+ XILINX_DIR=/
+ echo '        Xilinx directory is: //opt/Xilinx/'
        Xilinx directory is: //opt/Xilinx/
+ CONDA_DIR=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda
+ export PATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ '[' '!' -d /home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda ']'
+ conda config --add channels timvideos
Skipping channels: timvideos, item already exists
+ conda install binutils-lm32-elf
Fetching package metadata: ......
Solving package specifications: ..............
# All requested packages already installed.
# packages in environment at /home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda:
#
binutils-lm32-elf         2.25.1                        0  
+ conda install gcc-lm32-elf
Fetching package metadata: ......
Solving package specifications: ................
# All requested packages already installed.
# packages in environment at /home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda:
#
gcc-lm32-elf              4.9.3                         0  
+ conda install sdcc
Fetching package metadata: ......
Solving package specifications: ................
# All requested packages already installed.
# packages in environment at /home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda:
#
sdcc                      3.5.0                         0  
+ cd /home/shenki/tmp/HDMI2USB-misoc-firmware
+ git submodule update --recursive --init
+ git submodule foreach git submodule update --recursive --init
Entering 'third_party/fx2lib'
Entering 'third_party/liteeth'
Entering 'third_party/litescope'
Entering 'third_party/migen'
Entering 'third_party/misoc'
+ MIGEN_DIR=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen
+ cd /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen
+ cd vpi
+ export PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ python3 -c 'import migen'
+ MISOC_DIR=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc
+ cd /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc
+ cd tools
+ make
make: Nothing to be done for 'all'.
+ export PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc/tools/flterm --help
Serial boot program for MiSoC - v. 2.4
Copyright (C) 2007, 2008, 2009, 2010, 2011 Sebastien Bourdeauducq
Copyright (C) 2011 Michael Walle
Copyright (C) 2004 MontaVista Software, Inc

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, version 3 of the License.

Usage: flterm --port <port>
              [--speed <speed>] [--gdb-passthrough] [--debug]
              [--kernel <kernel_image> [--kernel-adr <address>]]
              [--cmdline <cmdline> [--cmdline-adr <address>]]
              [--initrd <initrd_image> [--initrd-adr <address>]]
              [--log <log_file>]

Default load addresses:
  kernel:  0x40000000
  cmdline: 0x41000000
  initrd:  0x41002000
+ python3 -c 'import misoclib'
+ LITEETH_DIR=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth
+ cd /home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth
+ true
+ export PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ python3 -c 'import liteeth'
+ MAKESTUFF_DIR=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff
+ '[' '!' -d /home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff ']'
+ cd /home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff
+ cd libs/libfpgalink
+ make deps
+ grep -E '^make'
make -C /home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader/ BRANCH=master deps
make[1]: Entering directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
make -C /home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer/ BRANCH=master deps
make[2]: Entering directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer'
make BRANCH=master
make[3]: Entering directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer'
make[3]: *** [lin.x64/dbg/.build/hexio.o] Error 1
make[3]: Leaving directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer'
make[2]: *** [deps] Error 2
make[2]: Leaving directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer'
make[1]: *** [/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer/lin.x64] Error 2
make[1]: Leaving directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
make: *** [/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader/lin.x64] Error 2
+ export LD_LIBRARY_PATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink/lin.x64/rel:
+ LD_LIBRARY_PATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink/lin.x64/rel:
+ export PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink/examples/python/:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ PYTHONPATH=/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink/examples/python/:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/liteeth:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/misoc:/home/shenki/tmp/HDMI2USB-misoc-firmware/third_party/migen:
+ python3 -c 'import fl'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink/examples/python/fl.py", line 69, in <module>
    cdll.LoadLibrary("libfpgalink.so")
  File "/home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda/lib/python3.4/ctypes/__init__.py", line 429, in LoadLibrary
    return self._dlltype(name)
  File "/home/shenki/tmp/HDMI2USB-misoc-firmware/build/conda/lib/python3.4/ctypes/__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libfpgalink.so: cannot open shared object file: No such file or directory
```

shenki, 2015-10-21:
Building manually:

```
shenki@icarus ~/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfpgalink :master 
$ make deps
make -C /home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader/ BRANCH=master deps
make[1]: Entering directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
make BRANCH=master
make[2]: Entering directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
mkdir -p lin.x64
echo '-I${ROOT}/libs/libfx2loader ' > lin.x64/incs.txt
mkdir -p lin.x64/dbg
(echo '-L${ROOT}/libs/libfx2loader/lin.x64/dbg -lfx2loader'; ) > lin.x64/dbg/libs.txt
mkdir -p lin.x64/dbg/.build
gcc -fPIC -g -D_DEBUG -c -m64 -DBYTE_ORDER=1234 -Wall -Wextra -Wundef -Wconversion -Wenum-compare -pedantic-errors -std=c99 -Wstrict-prototypes -Wno-missing-field-initializers -Wstrict-aliasing=3 -fstrict-aliasing -Warray-bounds  -I. -I/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/common  -I/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libusbwrap  -I/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libbuffer  -MMD -MP -MF lin.x64/dbg/.build/ram.o.d -Wa,-adhlns=lin.x64/dbg/.build/ram.o.lst ram.c -o lin.x64/dbg/.build/ram.o
ram.c:19:22: fatal error: liberror.h: No such file or directory
compilation terminated.
/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/common/top.mk:373: recipe for target 'lin.x64/dbg/.build/ram.o' failed
make[2]: *** [lin.x64/dbg/.build/ram.o] Error 1
make[2]: Leaving directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/common/top.mk:331: recipe for target 'deps' failed
make[1]: *** [deps] Error 2
make[1]: Leaving directory '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader'
/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/common/top.mk:410: recipe for target '/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader/lin.x64' failed
make: *** [/home/shenki/tmp/HDMI2USB-misoc-firmware/build/makestuff/libs/libfx2loader/lin.x64] Error 2
```

mithro, 2015-10-28:
Can you try again now? (Doing a fresh clone) We haven't changed anything but I think upstream may have?

CarlFK, 2015-10-28:
makestuff is being replaced with openocd, right?
thus don't bother building libfx2loader

mithro, 2015-11-03:
@shenki - We have now switched to openocd as the default programmer so I'm going to close this issue as obsolete.

closed 2015-11-03 by mithro