timvideos/HDMI2USB-litex-firmware #127: HDMI required to support 640x480@60

CarlFK, 2015-11-08:
I am not suggesting anything, just posting these comments. 
from:
http://logs.timvideos.us/%23timvideos/%23timvideos.2015-11-08.log.html#t2015-11-08T11:24:45

(10:49:55 AM) tumbleweed: CarlFK: no, it sees 640x480, even though the chromebook says it's sending 1024x768
(10:50:04 AM) tumbleweed: if I switch to mode 0, it works :(

(11:02:53 AM) tumbleweed: so, apparently all HDMI are required to support 640x480@60, which is why when the chromebook is given a mode it won't support, it went back to it
(11:03:02 AM) tumbleweed: (I just had an HDMI engineer come over and talk to me)

(11:03:21 AM) tumbleweed: he also suggests outputting a CEA-861 block https://en.wikipedia.org/wiki/Extended_Display_Identification_Data#EDID_1.3_data_format

(11:06:08 AM) tumbleweed: he said "ah, that's only 128 bytes, you should send the CEA-861 block too"
(11:06:15 AM) tumbleweed: matthew sweet, btw

mithro, 2015-11-09:
Related issues:
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/57 - Jay's whacky chrome book is still misbehaving
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/80 - Support resetting an video input without resetting the output
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/63 - Allow user to add custom EDID mode

mithro, 2017-03-19:
@joeladdison

joeladdison, 2017-03-19:
@mithro Ah yes, I know all about that. It won't be hard to add the extra block on to the end. 

We will just have to work out if we are going to follow the spec by supporting the fallback resolution, of if we just keep as is with how it is now and only display the one (or two) resolutions.