timvideos/HDMI2USB-litex-firmware #128: No (Black) output over usb and hdmi from input

sealne, 2015-11-10:
When connecting a pc via hdmi to the opsis the video input is not output to usb or hdmi and looks the same as when nothing is plugged in (tested by unplugging and re-plugging hdmi cable).

The PC seems to detect the opsis ok (didn't previously for me). video_mode was set to 9

xrandr --verbose

    HDMI-1 connected 1280x720+0+0 (0x26c) normal (normal left inverted right x axis y axis) 200mm x 112mm
            Identifier: 0x63
            Timestamp:  258868
            Subpixel:   unknown
            Gamma:      1.0:1.0:1.0
            Brightness: 1.0
            Clones:    
            CRTC:       1
            CRTCs:      0 1
            Transform:  1.000000 0.000000 0.000000
                        0.000000 1.000000 0.000000
                        0.000000 0.000000 1.000000
                       filter:
            EDID:
                    00ffffffffffff003d17545600000000
                    0019010380140bff0600000000000000
                    00000000000001010101010101010101
                    010101010101011d007251d01e20dc28
                    1400c8700000001e000000fc0048444d
                    493255534220310a2020000000100000
                    00000000000000000000000000000010
                    000000000000000000000000000000e5
            dithering depth: auto
                    supported: auto, 6 bpc, 8 bpc
            dithering mode: auto
                    supported: auto, off, static 2x2, dynamic 2x2
            scaling mode: Full
                    supported: None, Full, Center, Full aspect
            color vibrance: 150
                    range: (0, 200)
            vibrant hue: 90
                    range: (0, 180)
            underscan vborder: 0
                    range: (0, 128)
            underscan hborder: 0
                    range: (0, 128)
            underscan: off
                    supported: auto, off, on
      1280x720 (0x26c) 74.250MHz +HSync +VSync *current +preferred
            h: width  1280 start 1500 end 1540 total 1650 skew    0 clock  45.00KHz
            v: height  720 start  721 end  725 total  750           clock  60.00Hz

mithro, 2015-11-10:
Can you include a little bit more information about how this is connected up and what devices are in use?

sealne, 2015-11-10:
The PC is a Dell XPS M1530 laptop (about 8 years old) with its HDMI output connected to input0 on the opsis

sealne, 2015-11-16:
Still getting the same behaviour with latest master and running get-env.sh

mithro, 2015-11-16:
Gah, I got the wrong number in that commit message - should have been #126.

enjoy-digital, 2015-11-16:
sealne: what are you getting with "status" command?

sealne, 2015-11-16:
Interesting hadn't thought to check that. I was assuming that I was getting an input by the laptop thinking it saw the opsis connected and connecting the laptop to my monitor directly I do get an output

    [FLTERM] Starting...
    
    HDMI2USB>encoder on
    Enabling encoder
    HDMI2USB>output0 on
    Enabling output0
    HDMI2USB>video_matrix connect input0 output0
    Connecting input0 to output0
    HDMI2USB>video_matrix connect input0 encoder
    Connecting input0 to encoder
    HDMI2USB>status
    input0:  0x0
    input1:  0x0
    output0: 1280x720@60Hz from input0
    output1: 1280x720@60Hz from input0
    encoder: 1280x720 @ 0fps (0Mbps) from input0 (q: 50)
    ddr: read: 1760Mbps  write:    0Mbps  all: 1760Mbps

enjoy-digital, 2015-11-16:
Interesting, so it can be an issue in the FPGA with the input your laptop is generating or your laptop that is not generating video.

sealne, 2015-11-16:
Annoyingly that is my only pc that has HDMI out making testing harder

commit 006e5629 (2015-11-15): Use the legacy branch for lite* and mi* submodules.

Fixes #128 - Use legacy branch of Migen/MiSoC submodules.