timvideos/HDMI2USB-litex-firmware #132: Support video capture from Canon HV30 cameras

mithro, 2015-11-10:
Linux Australia owns 4 (maybe 5?) Canon HV30 cameras. We would like to use them with the HDMI2USB. Sadly they only 1080i60 on their HDMI output. The camera however *does* support 1080p24 and 1080p30 internally and will write this progressive format to the HDV tape using pulldown. This means that it is likely using a similar pulldown method to generate the HDMI signal.

Steps to proceed;
 - [ ] Confirm the HDMI output is actually progressive format with pulldown occurring.
 - [ ] Get the HDMI2USB accepting the interlaced video from the camera.
 - [ ] Get the HDMI2USB doing pulldown detection and deinterlacing.
 - [ ] Get the HDMI2USB doing 1080p to 720p downscaling.
 - [ ] Rejoice.

mithro, 2015-11-11:
http://eugenia.queru.com/2008/06/29/basic-canon-hv20hv30-knowledge/

mithro, 2015-11-11:
http://the-gadgeteer.com/2009/08/10/canon-vixia-hv30-hd-camcorder-review/

mithro, 2015-11-11:
* [Canon Page](https://www.usa.canon.com/internet/portal/us/home/support/details/camcorders/support-high-definition-camcorders/vixia-hv30/!ut/p/z1/hZBRT4MwEIB_yx54hF7A4fANmAsY0GwxE_tiOlbaGqBN6cD46-2mDy7qvLe7-y733SGMKoR7MgpGjJA9aW3-jMOXcp0tsjyF4mHl30JcPm6LMsgD2ITo6T8A2zb8ETGguxNwYd4a-LpMS4awIoa7om8kqmrS1VLvqR5QNRyUktq4XDDu7mkjenG0d78zo3gTxOVjANYYn--ETXQNcbgMs-g-gXw9_wFskysLRMs08FdH7y_g8lmslbvPD8b9LlhYf00bqqn2DtqWuTFquHHAgWmaPCYla6lXy86B30a4HAyqzkmkuuq9SKL8dd6ORTybfQBk2mBx/dz/d5/L2dBISEvZ0FBIS9nQSEh/)
 * http://gdlp01.c-wss.com/gds/4/0300000884/01/hv30nim-en.pdf

enjoy-digital, 2016-01-25:
HV30 is now working with Opsis (video_mode 12). It seems to always output 720x480 but probably in progressive for DV mode and pulldown for HD modes.

Not sure it will be easy to get HD from these cameras, do you really need 720p? (At least it will be too short to implement pulldown in hardware and doing it in software will probably not be possible: for that the encoder would need to be tightly synchronized with video input).

mithro, 2016-01-26:
@enjoy-digital Looks like your camera is currently in NTSC mode (`720x480@60Hz`) can you also test in PAL mode (you'll need the tool at http://nextdayvideo.com.au/HVhack.rar to switch your camera).

mithro, 2016-01-26:
@enjoy-digital After putting my camera into NSTC mode. I did managed to get some output from my HV30 but it is horrible mangled. https://goo.gl/photos/icLZAX5t7CnN12GD9

The status sometimes shows 720x0 or 720x something else, but not consistently.

Process for using the HVHack thing above;
 * Put your camera in play mode.
 * Slide the switch on right hand side too the little card icon (away from the tape icon).
 * Plug the camera into a Windows computer, it should appear as a USB device and the camera screen should go blue saying "PC connected".

mithro, 2016-01-26:
```
HDMI2USB>debug pll
framebuffer PLL:
0006 0008 0000 4400 1708 0097 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce 
dvisampler0 PLL:
0006 0008 0000 4400 1718 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce 
dvisampler1 PLL:
0006 0008 0000 4400 1708 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce 
```