timvideos/HDMI2USB-litex-firmware #133: Allow HDMI2USB devices to act as HDMI extenders via the Gigabit Ethernet port

mithro, 2015-11-10:
It would be good to allow two HDMI2USB devices be connected via their Gigabit Ethernet interfaces and have them act as HDMI extenders.

Gigabit Ethernet should be fast enough to transfer 720p60 video at raw pixel data (when using YUV encoding).

It would be nice if this was "over IP" but it would be okay if it was just over raw Ethernet frames.

You probably want to transfer the UART control data in a side channel too.

mithro, 2015-11-10:
This is different then the HDMI2Eth firmware as that is targeted at going HDMI2Eth<-Ethernet+IP->computer were as this is HDMI2USB<-Ethernet->HDMI2USB 

They might be able to share a lot of code though.

xfxf, 2015-11-10:
wow, this would be great - +1