timvideos/HDMI2USB-litex-firmware #134: Improve the issues found in CarlFK's video recording.

mithro, 2015-11-11:
http://5cda49ca88af98bf1f1e-b4c3b47b38bb1b572e0805ecabeeb59c.r76.cf2.rackcdn.com/21_06_36.mkv
http://5cda49ca88af98bf1f1e-b4c3b47b38bb1b572e0805ecabeeb59c.r76.cf2.rackcdn.com/20_42_44.mkv

```
[mjpeg @ 0x7feb7000ca60] Found EOI before any SOF, ignoring
[mjpeg @ 0x7feb7000ca60] No JPEG data found in image
```

[image]

[image]

[image]

mithro, 2015-11-11:
This was recorded with archive/v0.0.0-453-g4f2aac2/atlys/hdmi2usb from the prebuilt repo.

mithro, 2015-11-11:
Two similar issues from a [much older recording](https://www.youtube.com/watch?v=Bxkz6XSiB3Y) (also on a Atlys) 

https://usercontent.irccloud-cdn.com/file/0ggDsY1K/sypy2
https://usercontent.irccloud-cdn.com/file/L9QbzXiF/sypy1

The compression makes the problems much harder to see.

CarlFK, 2015-11-22:
Looks to be fixed.
tested with
unstable -> ../../archive/v0.0.0-465-g74f5a8f/atlys/hdmi2usb
firmware revision: 0607e926, built Nov 20 2015 20:12:01

[image]

mithro, 2015-11-22:
@enjoy-digital It does look like the issue was the patch you reverted at https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/74f5a8f8c023b2379c84cf61477a08d428de0192

enjoy-digital, 2015-11-23:
yes, for now I'm investingating on that in my fork.

mithro, 2017-03-19:
These should be fixed now.

closed 2017-03-19 by mithro