timvideos/HDMI2USB-litex-firmware #139: Updating liteEth to latest legacy commit fails due to FlipFlop removal

mithro, 2015-11-15:
FlipFlop seems to have been removed from liteEth in https://github.com/enjoy-digital/liteeth/commit/2b6dfa6a7e63297070b3aa5ed7fcc85dad375627

```
- make firmware (atlys hdmi2eth) (prerun)
---------------------------------------------
cd third_party/misoc && python3 make.py -X ../.. -t atlys_hdmi2eth -Ot firmware_filename ../../firmware/lm32/firmware.bin -Op programmer openocd -Ob ise_path /home/travis/build/timvideos/HDMI2USB-misoc-firmware/build/Xilinx/opt/Xilinx/ build-headers
Traceback (most recent call last):
  File "make.py", line 91, in <module>
    soc = top_class(platform, **top_kwargs)
  File "../../targets/atlys_hdmi2eth.py", line 120, in __init__
    self.submodules.encoder_streamer = UDPStreamer(convert_ip("192.168.1.15"), 8000)
  File "/home/travis/build/timvideos/HDMI2USB-misoc-firmware/gateware/streamer/__init__.py", line 21, in __init__
    self.submodules.level = level = FlipFlop(max=fifo_depth+1)
NameError: name 'FlipFlop' is not defined
make: *** [firmware-lm32] Error 1
```

enjoy-digital, 2015-11-15:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/403c33ec5f29346c8b4e715a8efe48c4e39d6678 should fix that.

mithro, 2015-11-16:
This has been fixed. I've updated the submodules to the latest legacy versions in https://github.com/timvideos/HDMI2USB-misoc-firmware/pull/138

closed 2015-11-16 by mithro