timvideos/HDMI2USB-litex-firmware #140: Add support for the miniSpartan6+ to HDMI2USB

mithro, 2015-11-15:
* Should be based on stuff at https://github.com/enjoy-digital/scarab-soc
 * The memory on the miniSpartan6+ is quite slow?
 * The USB interface can't emulate a real webcam

mithro, 2015-11-15:
@shenki has started work on this.

shenki, 2015-11-17:
Attempt to build BaseSoC to ensure we can program the ms6+: https://github.com/shenki/HDMI2USB-misoc-firmware/tree/minispartan-testing

WIP port: https://github.com/shenki/HDMI2USB-misoc-firmware/tree/minispartan6

mithro, 2016-01-10:
@cr1901 - See here for the miniSpartan6 stuff

mithro, 2016-01-10:
Patches for openocd can be found at http://openocd.zylin.com/#/c/3117/

cr1901, 2016-01-10:
OpenOCD JTAG works fine on miniSpartan6 using your patches. The flash proxy was broken last I checked however; powering the miniSpartan6 after loading data via the flash proxy does not execute the "flashed" bitstream.

Can someobody else who has a miniSpartan6 use OpenOCD and confirm this behavor w/ the flash proxy?

mithro, 2016-01-10:
@cr1901 I posted updated information about the flash proxy (the old information was incorrect). Can you please try with them?

mithro, 2016-04-16:
More work on this is available at https://github.com/mithro/HDMI2USB-misoc-firmware/tree/shenki-minispartan6%2B

mithro, 2016-04-17:
This was fixed in #249! Yay!

Still a couple of features are missing (see #255) but I'm closing this for the more specific issues.

closed 2016-04-17 by mithro