timvideos/HDMI2USB-litex-firmware #146: vizzini driver (Atlys) for Exar XR21V1414 USB UART is not reliable

xfxf, 2015-12-21:
The driver HDMI2USB-misoc-firmware installs appears to be buggy; 

* flterm appears to work fine (read/write)
* minicom/pyserial appear to only write to the device; trying to read from it causes the application to hang

@shenki may have a fix for this.

URL's of note:
 * https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=804350
 * https://github.com/shenki/linux/commits/vizzini

shenki, 2015-12-21:
@mithro and I tested ben's driver a few months back. We found that it worked for the most part, but we needed to ignore some of the USB descriptor. From the commit message:

```
usb-serial/vizzini: Fix probing for 1410 device

This device has the following configuration:

 device
  - interface
    - interrupt endpoint
  - interface
    - in bulk endpoint
    - out bulk endpoint

The first interface should be ignored in order to correctly probe. The
device appears to operate correctly, but this may be a horrible
```

Some further digging is required to work out what we miss out by ignoring that interrupt endpoint.

shenki, 2015-12-21:
I have sent the patch to the debian bug tracker

mithro, 2016-03-27:
@shenki ping :P