timvideos/HDMI2USB-litex-firmware #160: HDMI Audio to USB Sound

mithro, 2016-01-10:
_From @mithro on March 21, 2013 19:54_

http://www.usb.org/developers/docs/devclass_docs/audio10.pdf

_Copied from original issue: timvideos/HDMI2USB-jahanzeb-firmware#15_

mithro, 2016-08-19:
@rohit91 put together [this proposal](https://docs.google.com/document/d/1zbZyQyqNvs6Gd5bH8SCzUx8V1E7sVc_DchJI1uq9qlY/edit#) for GSoC in previous years. It might be useful starting point.

You'll also need the [fx2lib](https://github.com/mithro/fx2lib) and you'll probably want to use the Linux descriptor stuff that I'm working on for the UVC.

paddatrapper, 2016-08-19:
I'll start looking at the USB side of this. Waiting for FX2 board to arrive, which will take about a month. So I should be able to make some progress once it gets to me

mithro, 2017-08-04:
@paddatrapper Has been working on the USB side of things in https://github.com/timvideos/HDMI2USB-fx2-firmware/tree/master/audio