timvideos/HDMI2USB-litex-firmware #164: Design the system for auto-detection of Atlys VHDCI compatible expansion boards

mithro, 2016-01-10:
_From @mithro on July 17, 2014 6:47_

The firmware should be able to detect / understand what systems are connected to the VHDCI expansion port found on the Atlys and Numato-Conference boards. This is important as we get more expansion boards.

_Copied from original issue: timvideos/HDMI2USB-jahanzeb-firmware#57_

mithro, 2016-01-10:
See the following document for information on the various Atlys VHDCI expansion boards we know of https://docs.google.com/a/mithis.com/spreadsheets/d/1f-rBfR98f_ZZNIB7GFV79CGSHQNpj47uOU0JDwOJnV8/edit#gid=0

mithro, 2016-01-10:
See the following document for information on the auto-detection https://docs.google.com/a/mithis.com/document/d/1ZhDz50rve5UmnmyKynxySLHqfFuOwBPM7pYFoHo9uvs/edit

mithro, 2016-01-10:
See the following thread on the HDMI2USB mailing list on the contents of the EEPROM used for auto-detection - https://groups.google.com/forum/#!topic/hdmi2usb/q2WhYonDKjY

See the following thread on the HDMI2USB mailing list about VHDCI connector - https://groups.google.com/forum/#!topic/hdmi2usb/1d7sTZImeu8

mithro, 2016-01-10:
@rohit91 can you put a status update in this github issue? 

Are we at the stage where we can close this issue and open some more specific ones?

rohit91, 2016-01-11:
I guess we can keep this open. We hadn't finalized EEPROM contents. Does Milkymist TOFE expansion board have an EEPROM?

mithro, 2016-01-11:
https://hdmi2usb.tv/tofe/EEPROM.html

rohit91, 2016-01-11:
Oh my! That's quite detailed specification! Seems I was living in another world to have missed it. We can close this issue now definitely. More specific ones can be opened when needed.

mithro, 2016-01-12:
I only published it a couple of days ago :)

mithro, 2016-01-12:
We probably need some way of explaining what type of board, TOFE or VHDCI....

rohit91, 2016-01-12:
Wouldn't just changing the magic code to something like "VHDC\0" (again 5 bytes), be sufficient?