timvideos/HDMI2USB-litex-firmware #165: Output via HDMI2USB shows red/pink badly

xfxf, 2016-01-10:
Output to a monitor via HDMI2USB does not look the same as when connecting the device to the monitor directly.  In particular, red and pink look very bad.

Example of output routed via the HDMI2USB (source PC > HDMI2USB input, HDMI2USB output > monitor): https://drive.google.com/file/d/0B92_rpmFTXR5MUthbDMyeDFjR3M/view

Same output, but without HDMI2USB: https://drive.google.com/file/d/0B92_rpmFTXR5OEk3cWllaHgwQjQ/view

This same frame captured through HDMI2USB: https://usercontent.irccloud-cdn.com/file/Ysvg35MJ/Screenshot%20from%202016-01-10%2015-51-35.png

mithro, 2016-01-10:
@enjoy-digital Any idea what is going on here? 

At first I thought it was the RGB->YUV conversion stuff we did to reduce the memory bandwidth, but the effect looks more like pixel bleed or something?

Another though is that maybe it is sub-pixel ordering of the text? When connected directly, the subpixel ordering will be correctly reported to the computer, when connected via the HDMI2USB it will get the fake EDID which won't have that information in it?

mithro, 2016-01-10:
@xfxf This is a Windows computer as the source right? Do you know if you can change the sub-pixel order of text rendering under that at all?

Any chance you can get a picture which gets us the individual pixels, kinda like this -> http://farm4.staticflickr.com/3759/11989532244_30f3ecf68e_z.jpg

enjoy-digital, 2016-01-10:
It's probably related to RGB->YUV and 4:4:4 to 4:2:2 conversion.

We'll do more simulations of that during hackfest, we should be able to reproduce and fix the issue:
https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/gateware/csc/test

enjoy-digital, 2016-01-23:
@xfxf or someone else: I have a task for one of you:

If possible, I'd like that one of you test differents patterns/images/colors and try to see if the output seems fine or not. When the output seems to be a bit different or if you are able to reproduce the strange colors artifact there was on one video, can you post the original image here?

This would allow me to use these images in the YCbCr simulation and try to understand where are the possible rounding errors.

enjoy-digital, 2016-01-24:
Found the problem: https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/a6bf3bb410ecf53af84b65fc110a5023f558a220 with simulation:

Original:
[image]

Old (with bug):
[image]

New (fixed):
[image]

closed 2016-01-24 by enjoy-digital

mithro, 2016-01-24:
I can't actually see the difference in the bug / fixed version.

enjoy-digital, 2016-01-24:
Yes that's not the best picture for that.
I just tested on hardware, that's fine too.