timvideos/HDMI2USB-litex-firmware #170: Figure out a good way to share I2C code

mithro, 2016-01-20:
The HDMI2USB boards have a *lot* of I2C busses in them. Currently this means a lot of duplicated code. It would be good to refactor that code so it could be shared.

The I2C buses on the Atlys board are;
 * (slave) HDMI RX 1
 * (master) HDMI TX 1
 * (slave+master) Shared HDMI TX 2 + HDMI RX 2
 * (master) VHDCI auto detection interface
 * (master) Control of the IC on Rohit's VGA board

The I2C buses on the Opsis board are;
 * (slave) HDMI RX 1
 * (slave) HDMI RX 2
 * (master) HDMI TX 1
 * (master) HDMI TX 2
 * (slave+master) Opsis EEPROM + FX2 Boot loading
 * (master) TOFE Expansion EEPROM 
 * (master) Control of the IC on Rohit's VGA board

 they have -

enjoy-digital, 2016-01-22:
Is it fine for you?:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/404423e17169dfc7d44c8cfe2b16f72e4ccc555d
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/8a5f5835474ce39de8a41c9c88589ea20769c454
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/8464de6337587fadd05a6d688eb1a32350ff0523

mithro, 2016-01-23:
It's not a great solution, but it is a step in the right direction.

enjoy-digital, 2016-01-24:
Done via https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/2f9f8c3d0fed6e53d8b3424b1051c05d3c5da7e2

mithro, 2016-01-25:
@enjoy-digital has made a good start, but I think we can improve on this further - so I'm going to reopen it for now.

mithro, 2016-01-25:
Might be worth looking at http://www.i2cdevlib.com/