timvideos/HDMI2USB-litex-firmware #171: Fix csr_map so the numbers don't need to be manually incremented

mithro, 2016-01-20:
Currently, every time you add a new peripheral to the SoC, you need to manually increase the value of everything in the csr_map (see https://github.com/mithro/HDMI2USB-misoc-firmware/commit/e7fbe0db772d69e787017e2c8cc632559976e5ae for an example).

Computers are good at counting and auto-incrementing. We should change the csr_map to be automagic.

enjoy-digital, 2016-01-23:
This should do it:
https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/46750db5d2c1f9b40066bad83538660a5e9fbfca

enjoy-digital, 2016-01-23:
Done via https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/46750db5d2c1f9b40066bad83538660a5e9fbfca and https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/061421e37980c64c3b3d1546d47a47c0be200e21.

closed 2016-01-23 by enjoy-digital