timvideos/HDMI2USB-litex-firmware #172: Clean up all errors in the lm32 C code and then enable -Werror

mithro, 2016-01-23:
```
===========================
make -C firmware/lm32 all
make[1]: Entering directory `/home/travis/build/mithro/HDMI2USB-misoc-firmware/firmware/lm32'
 CC       isr.o
 CC       processor.o
processor.c: In function ‘fb_clkgen_write’:
processor.c:269:6: warning: variable ‘word’ set but not used [-Wunused-but-set-variable]
  int word;
      ^
processor.c: In function ‘edid_set_mode’:
processor.c:351:6: warning: unused variable ‘i’ [-Wunused-variable]
  int i;
      ^
processor.c:350:16: warning: unused variable ‘edid’ [-Wunused-variable]
  unsigned char edid[128];
                ^
 CC       hdmi_in0.o
 CC       hdmi_in1.o
 CC       pattern.o
 CC       edid.o
 CC       pll.o
pll.c: In function ‘program_data’:
pll.c:30:6: warning: unused variable ‘i’ [-Wunused-variable]
  int i;
      ^
pll.c: In function ‘pll_dump’:
pll.c:87:6: warning: unused variable ‘i’ [-Wunused-variable]
  int i;
      ^
 CC       ci.o
ci.c: In function ‘hdp_toggle’:
ci.c:313:6: warning: unused variable ‘i’ [-Wunused-variable]
  int i;
      ^
In file included from ci.c:7:0:
ci.c: At top level:
../../third_party/misoc/software/include/generated/sdram_phy.h:16:13: warning: ‘cdelay’ used but never defined
 static void cdelay(int i);
             ^
../../third_party/misoc/software/include/generated/sdram_phy.h:23:13: warning: ‘command_p1’ defined but not used [-Wunused-function]
 static void command_p1(int cmd)
             ^
../../third_party/misoc/software/include/generated/sdram_phy.h:28:13: warning: ‘command_p2’ defined but not used [-Wunused-function]
 static void command_p2(int cmd)
             ^
../../third_party/misoc/software/include/generated/sdram_phy.h:33:13: warning: ‘command_p3’ defined but not used [-Wunused-function]
 static void command_p3(int cmd)
             ^
../../third_party/misoc/software/include/generated/sdram_phy.h:67:13: warning: ‘init_sequence’ defined but not used [-Wunused-function]
 static void init_sequence(void)
```

enjoy-digital, 2016-01-23:
Done via https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/41ab6d96d5564d5c9458c1354dc6baad2eba8c87 and https://github.com/m-labs/misoc/commit/1330c7e68497827e7444c824594fb5485b32c799

enjoy-digital, 2016-01-23:
@mithro : that should be ok now, you just need to update misoc on travis-ci.

enjoy-digital, 2016-01-23:
@mithro : and then you can activate -Werror

mithro, 2016-04-18:
FYI - @Vikram9866 

If -Werror doesn't return any errors when building the firmware now (it looks like Florent may have fixed the existing errors already?), you just need to send a pull request which enables it the flag.

Make sure that you check all the different target / board types ("export BOARD=xx" and "export TARGET=xx"). You can see the valid board / target combinations at https://travis-ci.org/timvideos/HDMI2USB-misoc-firmware/builds/123648739

The next logical step is to enable -Wall and fixing the errors from that (might be a good idea to do that before enabling -Werror).