timvideos/HDMI2USB-litex-firmware #180: Test pattern failure - black/white alternating lines

mithro, 2016-01-24:
Sending a test pattern into the HDMI2USB which is just alternatives white/black pixels you get lots of pixels which are black which shouldn't be. The pixels change every time - this is pretty visually distinctive when watching it in mplayer.

Taking a screenshot you get the following;
[image]

mithro, 2016-01-24:
Closer zoomed in image;

[image]

enjoy-digital, 2016-01-24:
Can you share your orignal pattern?

mithro, 2016-01-24:
It's been generated by a hardware device but we don't see any of the artifacts on the monitor.

I'll try and replicate with a computer....

mithro, 2016-01-25:
I've created a whole bunch of test patterns in the following repository -> https://github.com/timvideos/test-patterns

I then confirmed that when running the lines input through a software jpeg encoder (the one that is part of imagemagick), it doesn't produce the pixel artifacts (with q=50,q=75,q=85,q=100).

enjoy-digital, 2016-02-03:
This seems to be an issue inside the encoder. It's probably:
- input that is not handled correctly in some case.
- backpressure from the output that is not propagated correctly.

We will need simulations to fix this. Another option can also be to create our encoder in Migen.

mithro, 2017-05-19:
@enjoy-digital I'm pretty sure this is fixed now, right?

enjoy-digital, 2017-05-19:
@mithro: yes it was fixed during the hackfest before LCA. That was not a problem in the encoder but in the module just before it.

closed 2017-05-19 by enjoy-digital