timvideos/HDMI2USB-litex-firmware #184: Opsis board does not recognise XA20 camera

xfxf, 2016-01-24:
This has been tested + confirmed multiple times.

My Canon XA20 cameras outputting 720p50 works as expected with an Atlas board using firmware compiled using commit 480cd42c91ffdae32b7f785a0a8c9ac11684311e.

It does not work with an Opsis running shipping firmware.  Input is never detected (XA20 is input0):

input0:  0x0
input1:  1280x720
output0: 1280x720@50Hz from input0
output1: 1280x720@50Hz from pattern
encoder: 1280x720 @ 0fps (0Mbps) from pattern (q: 50)
ddr: read: 1468Mbps  write:  884Mbps  all: 2352Mbps

Tried using multiple Opsis's, both inputs, different cables.

mithro, 2016-01-24:
What is the revision on the Opsis?

Is the status from the Atlys or Opsis? How does it compare to the other?

micolous, 2016-01-24:
Status is from the Opsis.  input1 has a signal generator attached signal generator.  Tests were attempted in mode 9 (720p60) and 10 (720p50) with no difference.

With the Atlys, the input is shown as normal input, at 1280x720, and works.

With the Opsis, it is shown as 0x0 and does not work.

mithro, 2016-01-25:
The same behaviour with both inputs?

Can you get the full status output and what the camera is menu is showing in the following cases;
 * {Atlys,Opsis}
   * {720p60 mode, 720p50 mode}
     * XA20 on Input 0
     * XA20 on Input 1

xfxf, 2016-01-27:
Instructions are as follows (both tests, only difference is Atlys or Opsis):

* Plug TV into output0
* Plug XA20 into input0, set to 720p
* Turn on HDMI2USB
* (for Atlys) flash firmware
* Connect to firmware
* video_mode 10
* video_matrix connect input0 output0
* status on
* Turn off XA20, turn it back on again (Atlys does not see device otherwise - must be done AFTER video_mode is set)

Atlys board (works):

HDMI2USB>debug pll
framebuffer PLL:
0006 0008 0000 4400 1708 0017 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce
dvisampler0 PLL:
0006 0008 0000 4400 1708 0006 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce
dvisampler1 PLL:
0006 0008 0000 4400 1718 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce
HDMI2USB>status
input0:  0x0
input1:  1280x720
output0: 1280x720@50Hz from input1
output1: 1280x720@50Hz from input0
encoder: off
ddr: read: 1470Mbps  write:  737Mbps  all: 2207Mbps

Opsis (doesn't work, never shows anything on input0, even when power cycling):
HDMI2USB>status
input0:  0x0
input1:  0x0
output0: 1280x720@50Hz from input0
output1: 1280x720@50Hz from pattern
encoder: 1280x720 @ 0fps (0Mbps) from pattern (q: 50)
ddr: read: 1468Mbps  write:    1Mbps  all: 1469Mbps
HDMI2USB>debug pll
framebuffer PLL:
0006 0008 0000 4400 1708 0097 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce
dvisampler0 PLL:
0006 0008 0000 4400 1718 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce
dvisampler1 PLL:
0006 0008 0000 4400 1708 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce

Camera menu from above (note, camera cannot be switched from 720p60 to 720p50 mode - it's rare for any camera to do this without firmware hacks, assuming identical hardware.  I have assumed this is switching the video_mode on HDMI2USB).

In each case, after plugging the XA20 in and setting video_mode, I turn off and on the XA20.

Atlys, 720p60, input0: ---- (indicates no output)
Atlys, 720p60, input1: can't test, don't have mini HDMI > micro HDMI cable
Atlys, 720p50, input0: DVI 1280x720P
Atlys, 720p50, input1: can't test, don't have mini HDMI > micro HDMI cable

Opsis, 720p60, input0: ---- (indicates no output)
Opsis, 720p60, input1: ---- (indicates no output)
Opsis, 720p50, input0: DVI 1280x720P
Opsis, 720p50, input1: DVI 1280x720P

xfxf, 2016-01-27:
I plug the XA20 into Tim's Digitech 2-way HDMI splitter, plug it into the Opsis, set video_mode to 10 (720p), power cycle the splitter/XA20, and now it works.

HDMI2USB>status
input0:  1280x720
input1:  0x0
output0: 1280x720@50Hz from input0
output1: 1280x720@50Hz from pattern
encoder: 1280x720 @ 0fps (0Mbps) from pattern (q: 50)
ddr: read: 1472Mbps  write:  736Mbps  all: 2208Mbps
HDMI2USB>debug pll
framebuffer PLL:
0006 0008 0000 4400 1708 0097 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce 
dvisampler0 PLL:
0006 0008 0000 4400 1718 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce 
dvisampler1 PLL:
0006 0008 0000 4400 1708 008e 0501 8288 4201 0d90 00a1 0111 1004 2028 0802 2800 0288 8058 020c 0200 1210 400b fc21 0b21 7f5f c0eb 472a c02a 20b6 0e96 1000 d6ce