timvideos/HDMI2USB-litex-firmware #185: Embed a wide range of test patterns

mithro, 2016-01-25:
Test patterns are really useful for doing multiple things;
 * Testing the HDMI2USB is working.
 * Testing the device the HDMI2USB is connected too. (Things like overscan, underscan, aspect ratio issues.)

We should add a whole bunch of these useful test patterns to the firmware.

For example test patterns, see https://github.com/timvideos/test-patterns

mithro, 2016-01-25:
Generating static patterns is easy on the lm32 (patterns which use the same value for every frame).

Generating dynamic patterns is harder. The lm32 doesn't have the speed to generate them dynamically. The two options are;
  * Pre-generate all the frames into ram and then just cycle the frame pointers.
  * Create some HDL which can generate the patterns. This could be reused for doing dynamic wipes for the [hardware mixing](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/27). See [mask generator](https://docs.google.com/document/d/1ZjM1Brrks0lg1CJp2Rt1BH8-MhJamrKUeCUB4s4nzoA/edit#heading=h.yxbibecdjax2).

CarlFK, 2017-11-26:
On the current test pattern (color bars) add a square and/or circle for checking aspect ratio of the downstream processing (display device (projector) and whatever hardware is in front of it and  the mjpg stream  that eventually ends up on a desktop media player.)

mithro, 2017-12-03:
Pattern generation code is found at https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/firmware/pattern.c#L167-L246

mithro, 2017-12-03:
```c
void pattern_fill_framebuffer(int h_active, int w_active)
{
	int i, j;
	int color;
	flush_l2_cache();
	color = -1;
	volatile unsigned int *framebuffer = (unsigned int *)(MAIN_RAM_BASE + PATTERN_FRAMEBUFFER_BASE);
	if(pattern == COLOR_BAR_PATTERN) {
		/* color bar pattern */
		for(i=0; i<h_active*w_active*2/4; i++) {
			if(i%(h_active/16) == 0)
				color = inc_color(color);
			if(color >= 0)
				framebuffer[i] = color_bar[color];
		}
	} else {
		/* vertical black white lines */
		for(i=0; i<h_active*w_active*2/4; i++) {
			if(i%(h_active/16) == 0)
				color = inc_color(color);
			if(color >= 0)
				framebuffer[i] = 0x801080ff;
		}
	}

	// draw a border around that.
	for (i=0; i<h_active*2; i++) {
		framebuffer[i] = YCBCR422_WHITE;
	}
	
	for (i=(w_active-4)*h_active*2/4; i<h_active*w_active*2/4; i++) {
		framebuffer[i] = YCBCR422_WHITE;
	}
	
	for (i=0; i<w_active*2; i++) {
		// do the left bar
		for (j=0; j<2; j++) {
			framebuffer[(i*h_active)+j] = YCBCR422_WHITE;
			framebuffer[(i*h_active)+j + (1*h_active/2)] = YCBCR422_WHITE;
		}
		
		// do the right bar
		for (j=h_active-2; j<h_active; j++) {
			framebuffer[(i*h_active)+j] = YCBCR422_WHITE;
			framebuffer[(i*h_active)+j + (1*h_active/2)] = YCBCR422_WHITE;
		}		
	}

	// Line 1 - uptime + version information
	int line = 1;
	pattern_draw_text(19, line, (char*)git_describe);
	line++;
	// Line 2 - build time data
	pattern_draw_text(1, line, "Built: "__DATE__" "__TIME__);
	line++;
	// Line 3- running on
	pattern_draw_text(1, line, "Running on ");
	pattern_draw_text(12, line, (char*)board);
	line++;

	// Line 4 - empty
	line++;

#ifndef HIDE_ADVERT
	// Line 5 - Hi!
	pattern_draw_text(1, line, "Hi! I am HDMI2USB ");
	line++;
	// Line 6+7 - Want...
	pattern_draw_text_color(1, line, "Want to hack on FOSS video capture systems?", YCBCR422_BLUE, YCBCR422_WHITE);
	line++;
	pattern_draw_text_color(1, line, "Get in touch with us! #timvideos on Freenode IRC", YCBCR422_RED, YCBCR422_WHITE);
	line++;
	// Line 8 - URLs..
	pattern_draw_text(1, line, "code.timvideos.us / enjoy-digital.fr");
	pattern_draw_text_color(6, line, "tim", YCBCR422_WHITE, YCBCR422_RED);
	pattern_draw_text_color(9, line, "videos", YCBCR422_WHITE, YCBCR422_BLUE);
	pattern_draw_text_color(27, line, "digital", YCBCR422_WHITE, YCBCR422_CYAN);
#endif

	flush_l2_cache();
```

andrewreds, 2018-02-16:
I'm thinking of picking up this issue.

Plan being:

- [ ] Add circles & squares to the bars image (for aspect ratio checking)
- [ ] Add horizontal lines image (or make the vertical lines pattern be split screen with a horizontal line pattern)
- [ ] Add more debug info (eg, the "status short" command)
- [ ] Add support for hardware swapping between two images in pattern mode (I think this can be done without gateware changes)
- [ ] Add in contrast/brightness images (doing very dark/light colour bars that flash on & off)
- [ ] Add in drop frame check (flash an image with it's inverse @ 60 fps, on a 50% grey background)
- [ ] Add in motion blur test (moving box of a solid colour. By my very rough calcus, we should be able to push ~ 20k pxs per frame. Enough to draw a couple of moving boxs @ 60 fps)
- [ ] Add in beep test (for audio sync once it comes)
- [ ] Add support for selecting a specific test from UART
- [ ] Add support for adding a colour filter to all patterns from UART (for testing just one colour channel)
- [ ] Add ability to auto cycle through tests

Any other tests I should be adding?

mithro, 2018-02-16:
We have a bunch of test patterns we have found very useful here -> https://github.com/timvideos/test-patterns

Definitely start with the static patterns, doing non-static patterns is going to be a lot more complicated (as the CPU isn't fast enough to generate patterns in real time).

CarlFK, 2018-02-16:
for unit testing, it would be handy to be able to not show the text.

I expect there will also be tests that look for the text, but being able to
turn it off would be handy.

the hard part about this is  coming up with the command name.  ;)