timvideos/HDMI2USB-litex-firmware #209: ISE generates both working and non-working firmware from the same input verilog

mithro, 2016-02-03:
ISE is currently generating both working and non-working firmware from the same git commit to the repo.

Version v0.0.0-606-gebbfdc5 was built twice on Travis, the [first version](https://github.com/timvideos/HDMI2USB-firmware-prebuilt/blob/9550b0bfbeaba69346481058bbc07ff00f6db32a/archive/lca2016/v0.0.0-606-gebbfdc5/atlys/hdmi2usb/atlys_hdmi2usb-hdmi2usbsoc-atlys.bit) fails to produce valid JPEG frames, the [second version](https://github.com/timvideos/HDMI2USB-firmware-prebuilt/blob/master/archive/lca2016/v0.0.0-606-gebbfdc5/atlys/hdmi2usb/atlys_hdmi2usb-hdmi2usbsoc-atlys.bit) kind of works.

mithro, 2016-02-03:
The current theory is that we are missing timing constraints needed to make sure the encoder (or something related to the encoder) to no longer work.

mithro, 2016-02-03:
This issue might be related to the following issues;
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/77 - DDR Memory is currently broken on the Opsis 
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/5 - Attribute CLKIN_PERIOD (on DCM_CLKGEN_1) doesn't match PERIOD constraint?
 * https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/210 - Timing constraints not propagating through DCM/PLLs in the firmware

mithro, 2016-03-21:
@jordens created the following LD_PRELOAD to try and make ISE more deterministic. https://github.com/jordens/unrandom

jordens, 2016-03-21:
Credit should go to @whitequark.

mithro, 2016-03-21:
Random comment I found from https://forums.xilinx.com/t5/Welcome-Join/Multiple-Seed-Place-amp-route/td-p/503024

> @markcurry is correct - there is no mechanism in Vivado to do the equivalent of seeds (which is what the "Cost Tables" in ISE were). But the removal of the seeds (Cost Tables) from Vivado wasn't based on a decision as to its usefulness...

See `Starting Placer Cost Table (1-100)` in http://www.xilinx.com/itp/xilinx10/isehelp/pp_db_place_and_route_properties.htm

See also https://www.fpgarelated.com/showthread/comp.arch.fpga/46386-1.php

mithro, 2016-03-25:
A fix for the prj file generation can be found in https://github.com/timvideos/migen/commit/c468315143d44a21f15e5353c3a9c047d3aa5364

mithro, 2016-03-25:
With the current issue in #227 you get the following
```
S ./third_party/misoc/build/.keep_me
S ./third_party/misoc/build/_xmsgs/bitgen.xmsgs
S ./third_party/misoc/build/_xmsgs/map.xmsgs
S ./third_party/misoc/build/_xmsgs/ngdbuild.xmsgs
S ./third_party/misoc/build/_xmsgs/par.xmsgs
S ./third_party/misoc/build/_xmsgs/trce.xmsgs
D ./third_party/misoc/build/_xmsgs/xst.xmsgs
S ./third_party/misoc/build/build_opsis_hdmi2usb-hdmi2usbsoc-opsis.sh
S ./third_party/misoc/build/edid_mem.init
S ./third_party/misoc/build/edid_mem_1.init
S ./third_party/misoc/build/header.hex
D ./third_party/misoc/build/mem.init
D ./third_party/misoc/build/mem_2.init
D ./third_party/misoc/build/netlist.lst
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.bgn
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.bin
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.bit
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.bld
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.drc
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ncd
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ngc
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ngc_xst.xrpt
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ngd
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.pad
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.par
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.pcf
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.prj
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ptwx
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.srp
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.tsi
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.twr
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.twx
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ucf
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.unroutes
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.v
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.xpi
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.xst
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_bitgen.xwbt
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_map.map
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_map.mrp
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_map.ncd
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_map.ngm
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_ngdbuild.xrpt
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_pad.csv
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_pad.txt
S ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_summary.xml
D ./third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis_usage.xml
D ./third_party/misoc/build/par_usage_statistics.html
S ./third_party/misoc/build/top.lso
D ./third_party/misoc/build/top_map.xrpt
D ./third_party/misoc/build/top_par.xrpt
S ./third_party/misoc/build/webtalk.log
S ./third_party/misoc/build/xlnx_auto_0_xdb/cst.xbcd
D ./third_party/misoc/build/xst/work/work.sdbl
D ./third_party/misoc/build/xst/work/work.sdbx
D ./third_party/misoc/build/xst/work/work.vdbl
D ./third_party/misoc/build/xst/work/work.vdbx
```

S == files that are the same
D == files that are different

The mem.init and mem_3.init are different because of #228

mithro, 2016-03-25:
[This PDF contains the command line arguments for the Xilinx tools](http://www.xilinx.com/support/documentation/sw_manuals/xilinx14_7/devref.pdf) which might be useful for figuring out what is going on.

mithro, 2016-03-25:
The output of the first step is the ngc file (`third_party/misoc/build/opsis_hdmi2usb-hdmi2usbsoc-opsis.ngc`) which is different, so that would be the first thing to fix.

closed 2019-08-02 by mithro